ElectronicCats/mpu6050 #3: library issues

cookins, 2019-06-20:
hello
got some issue with code\library -
im getting 'class MPU6050' has no member named 'begin' in this sketch:
`#include <Wire.h>
#include <MPU6050.h>
 
MPU6050 mpu;
 
void setup()
{
Serial.begin(9600);
 
Serial.println("Initialize MPU6050");
while(!mpu.begin(MPU6050_SCALE_2000DPS, MPU6050_RANGE_2G))
{
Serial.println("Error");
delay(1000);
}
pinMode(13, OUTPUT);
 
mpu.setThreshold(2);
}
 
void loop()
{
Vector rawGyro = mpu.readRawGyro();
Vector normGyro = mpu.readNormalizeGyro();
float normal_x = normGyro.XAxis;
float normal_y = normGyro.YAxis;
float normal_z = normGyro.ZAxis;
Serial.print(" Xnorm = ");
Serial.print(normal_x);
Serial.print(" Ynorm = ");
Serial.print(normal_y);
Serial.print(" Znorm = ");
Serial.println(normal_z);
 
if(abs(normal_x) + abs(normal_y) + abs(normal_z) > 0.2){
digitalWrite(13, HIGH);
}else{
digitalWrite(13, LOW);
}
 
delay(10);
}`

if you have any hints - will be happy. thanks

sabas1080, 2019-06-21:
the begin is 

while(!**mpu.begin(MPU6050_SCALE_2000DPS, MPU6050_RANGE_2G**))

cookins, 2019-06-22:
im not programmer, but can you write me in details where is mistake?
its not my sketch, but author have it working he say...

as i understand setup cant be start from "while"?

EllieOwen, 2019-06-22:
The MPU6050 library that is located at C:\Users\UserName\Documents\Arduino\libraries\MPU6050\src\MPU6050.h doesn't have a begin method.  try 
Serial.println("Initialize MPU6050");
mpu.initialize();
while(!mpu.testConnection())
{
Serial.println("Error");
delay(1000);
}

You will then find there is also no setThreshold() method.
Are you sure this MPU6050.h library is the correct one or is is there a customized one you need to include?  Is there another MPU6050.h file in the sketch folder?  If so try
#include "MPU6050"

gonzalosalvador, 2019-09-05:
Hi, I think there's something wrong about accel scaling.
Regarding lines 1818-1821 of MPU6050.cpp, it says:

* AFS_SEL | Full Scale Range | LSB Sensitivity
 * --------+------------------+----------------
 * 0       | +/- 2g           | 8192 LSB/mg
 * 1       | +/- 4g           | 4096 LSB/mg
 * 2       | +/- 8g           | 2048 LSB/mg
 * 3       | +/- 16g          | 1024 LSB/mg

However, in the Invensense datasheet appears to be something like:

* AFS_SEL | Full Scale Range | LSB Sensitivity
 * --------+------------------+----------------
 * 0       | +/- 2g           | 16384 LSB/mg
 * 1       | +/- 4g           | 8192 LSB/mg
 * 2       | +/- 8g           | 4096 LSB/mg
 * 3       | +/- 16g          | 2048 LSB/mg

Plus thinking about it, being int16_t, max number should be 32768 so if range it's 2g, 1g should be half of it (16384).

Also I've experienced the curious event of calling getAcceleration and obtaining around 0 0 16384, but when activating DMP and calling

mpu.dmpGetQuaternion(&dmp_q, mpuFifoBuffer);
mpu.dmpGetGravity(&dmp_gravity, &dmp_q);
mpu.dmpGetYawPitchRoll(dmp_ypr, &dmp_q, &dmp_gravity);
mpu.dmpGetAccel(&dmp_aa, mpuFifoBuffer);
mpu.dmpGetLinearAccel(&dmp_aaReal, &dmp_aa, &dmp_gravity);

this dmp_aa says around 0 0 8192. Being accel configured at 2g at all times.
Looking through the Internet I bumped into this Arduino forum thread talking about the same thing but not coming really into a conclusion (https://forum.arduino.cc/index.php?topic=176064.0).

Also dmpGetGravity appears to substract the gravity component but the fixed number of 8192.

Though I might be doing something wrong, I really think there's some general missunderstanding with the acceleration scales in the library.
Hope it helps you allmighty allknowy guys revise it and maybe fix it or just explain it better. Thanks a lot!
PS: I've gone through every single line of code when initializing/configuring/calibrating/using the MPU/DMP so that's why I didn't put any line of code here because I really think it's about the library

sabas1080, 2019-09-05:
Hi @gonzalosalvador 

we have updated the library solving some problems, please update and test.

And please open a new issue of github

sabas1080, 2019-09-05:
Close issue for error in class

closed 2019-09-05 by sabas1080