ElectronicCats/mpu6050 #7: Bits operation mistake

mateusmenezes95, 2020-04-26:
First, thank you for maintaining this library!

So, I'm adapting this library to work in the raspberry pi zero. I'm reading the whole code to understand better how it works and I stopped in the methods that read specific bits of a byte or word read from I2C device, as below:

https://github.com/ElectronicCats/mpu6050/blob/4e0f3a2dba28aa48691832dacf7049004efae22d/src/I2Cdev.cpp#L139-L153

I was in doubt if these methods work as expected and I made some tests. The code that I used is very simple as can be seen below
```cpp
#include <iostream>
#include <cstdint>
#include <bitset>

#define BYTE_TO_BINARY_PATTERN "%c%c%c%c%c%c%c%c\n"
#define BYTE_TO_BINARY(byte)  \
  (byte & 0x80 ? '1' : '0'), \
  (byte & 0x40 ? '1' : '0'), \
  (byte & 0x20 ? '1' : '0'), \
  (byte & 0x10 ? '1' : '0'), \
  (byte & 0x08 ? '1' : '0'), \
  (byte & 0x04 ? '1' : '0'), \
  (byte & 0x02 ? '1' : '0'), \
  (byte & 0x01 ? '1' : '0')

int main() {
  uint8_t byte_read = 0b01101001;
  uint8_t bit_start = 4;
  uint8_t length = 3;

  printf("Raw val  = " BYTE_TO_BINARY_PATTERN, BYTE_TO_BINARY(byte_read));

  uint8_t mask = ((1 << length) - 1) << (bit_start - length + 1);
  printf("Raw mask = " BYTE_TO_BINARY_PATTERN, BYTE_TO_BINARY(mask));

  byte_read &= mask;
  printf("byte_read & mask = " BYTE_TO_BINARY_PATTERN, BYTE_TO_BINARY(byte_read));

  byte_read >>= (bit_start - length + 1);
  printf("byte_read shifted right by %d = " BYTE_TO_BINARY_PATTERN, bit_start - length + 1, BYTE_TO_BINARY(byte_read));
}
```
Using the same example that is in the body of the method, the result was this:
```console
Raw val  = 01101001
Raw mask = 00011100
byte_read & mask = 00001000
byte_read shifted right by 2 = 00000010
```
That is different of the correct result: `011` (bits, 4, 5, and 6 of `01101001`) 

Besides that, a strange behavior happens when e.g., `bit_start = 1` and `length = 5` because the result of this operation `(bit_start - length + 1)` will be negative what doesn't exist in shift operations. The output of this scenario is this:
```console
Raw val  = 01101001
Raw mask = 00000000
byte_read & mask = 00000000
byte_read shifted right by -3 = 00000000
```
I'm trying to make a better approach and if you want I can share with you.

And finally, do you think that this operation does not affect the  MPU6050 class?

sabas1080, 2020-04-26:
Hi @mateusmenezes95 
Do you are testing in Arduino or Raspberry Pi?

This Library is Arduino, when porting to Raspberry here may be differences, Can you try in arduino?

readBits if used in MPU6050

https://github.com/ElectronicCats/mpu6050/blob/d688ba3933fd6b3226683e6ac3c34c7bd54340cb/src/MPU6050.cpp#L159

Finally welcome a pullrequest with your solution :D

mateusmenezes95, 2020-04-26:
Hello @sabas1080 

I did some tests in Raspberry only changing the main I2C methods (read and write bytes e words) and I got consistent values from MPU6050. The doubt appears when I was optimizing the code to fit with Raspberry capabilities. But I think that my points don't depend on platforms since they are about bit operations.

I'll do more tests in the Raspberry board and if it works I'll try to do the changes in this library

sabas1080, 2020-04-26:
@mateusmenezes95 it seems good to me to compare, thanks

mateusmenezes95, 2020-04-26:
@sabas1080 Finally I understood what happens in bit operation :smile:

The bit mask is applied from left to right. So, if the `bit_start = 4` and the `length = 3`, the mask will be applied in the bits 4, 3 and 2 and not in bits 4, 5 and 6 that I was thinking.

In a brief, I think the problem was merely a convention. Then, there isn't the need to change this library.

I apologize for the confusion!

closed 2020-04-26 by mateusmenezes95