ElectronicCats/mpu6050 #12: Library does not interface with MPU6050 on Heltec Wifi V2 ESP32 Board

tonyvr4, 2020-08-09:
I am trying to use this library to interface with an MPU-6050 accel/gyro on a Heltec ESP32 WiFi V2 Development board

The board pinout shows SDA as Pin 21 and SCL as Pin 22

Upon compile I get this message
WARNING: library MPU6050 claims to run on avr, samd architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).

sabas1080, 2020-08-09:
Hi @tonyvr4 

I have added support for ESP32, can you test?

tonyvr4, 2020-08-09:
Still fails to initialize the MPU6050.  I have tried the IMU-Zero, MPU6060_DMP6

The MPU6050_DMP6_ESPWiFi will not even run.  It seems that I do not have most of those libraries for the Heltec ESP32

The compile stops with MPU6050_DMP6_ESPWiFi:51:24: fatal error: OSCMessage.h: No such file or directory

tonyvr4, 2020-08-10:
Progress.  I was able to get the MPU6050 to initialize.  Also the update you posted now removed the compile error.

I don't know if this will work on any pin but I will have to test that after I get the interrupt to work.

I do not know how to setup and use a pin as an Interrupt on the ESP32.  The accel/gyro generates an interrupt when data is available.  Then I can view the readings when I figure this out.

Here is what I did so far:
I moved the MPU6050 to the first I2C bus, the one that the display is connected to. 
This is SDA-Pin4 and SCL-Pin 15

The steps are as follows:
1) Load the heltec library
#include "heltec.h"

2) Now you can enable the built-in OLED display with this line
Heltec.begin(true /*DisplayEnable Enable*/, false /*LoRa Enable*/, true /*Serial Enable*/);

3) The key to getting I2C to work seems to be that now you have to initialize the display then other devices can be used
Heltec.display -> display();

4) It is time to start the wire.  It defaults to SDA-Pin4 and SCL-Pin 15
Wire.begin();

5) Now you can initialize the MPU6050
mpu.initialize();

The next step would be to enable the interrupt.  On the Arduino Uno all you have to do is plug the MPU6050 INT pin into Uno pin 2 and it worked right away.

But I am not sure how to setup and use a GPIO pin as an Interrupt on the Heltec  board

sabas1080, 2020-08-17:
solved according to https://github.com/ElectronicCats/mpu6050/issues/13

closed 2020-08-17 by sabas1080