ElectronicCats/mpu6050 #13: Is it possible to use DMP6 example code without using Interrupt Pin ?

tonyvr4, 2020-08-17:
I am looking to use your awesome lib in a data logger project to write GPS data and Acceleration data to an SD card.
The microprocessor is an ESP32 Heltec development board.

Your example MPU6050_DMP6 works great!
However the interrupt causes issues using GPS data.  

Is it possible to use the DMP6 code and eliminate the interrupt component?
This way I will read and store acceleration values in the main program loop and write the data to SD along with the GPS data

Thanks for your help

sabas1080, 2020-08-17:
Hi @tonyvr4 

The interrupt isn´t exclusive of this library, attachInterrupt is of arduino 
 https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino#L225

only desactive attachInterrupt and call the function "dmpDataReady"  for time or change the value of "mpuInterrupt" 
https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino#L154

closed 2020-08-20 by sabas1080

tonyvr4, 2020-09-05:
I have been testing the use of the MPU6050 without interrupt in my GPS-Accelerometer Data logger. 
I had to add a line to reset the FIFO before reading the FIFO since it was giving corrupted data.

For some reason it gets stuck in this loop on occasion, I have not determined why.  The code was from the DMP6 example

    // wait for MPU extra packet(s) available
    while (fifoCount < packetSize) {
        if (fifoCount < packetSize) {
          // try to get out of the infinite loop
          Serial.println("In FIFO Loop");  //for troubleshooting
          fifoCount = mpu.getFIFOCount();
        }  
    }