ElectronicCats/mpu6050 #15: how to get Temperature from MPU6050

tonyvr4, 2020-08-24:
How can I get the Temperature from the MPU6050 ?

Thanks

sabas1080, 2020-08-24:
You can use 
getTempSensorEnabled() https://github.com/ElectronicCats/mpu6050/blob/d688ba3933fd6b3226683e6ac3c34c7bd54340cb/src/MPU6050.cpp#L2491

and getTemperature()
https://github.com/ElectronicCats/mpu6050/blob/d688ba3933fd6b3226683e6ac3c34c7bd54340cb/src/MPU6050.cpp#L1870

tonyvr4, 2020-08-24:
If I use   mpu.setTempSensorEnabled(true);

and read the temperature with mpu.getTemperature()

I am getting bizarre readings like 4504 or 4494 instead of a human readable temperature

sabas1080, 2020-08-24:
Oh, ok, you have to divide the temperature reading by 340 and then add +36.53.  The result is Celsius.

`float temperature = (rawTemp / 340.0) + 36.53;`

tonyvr4, 2020-09-05:
Sorry for the delay.  This worked.  Thanks!

closed 2020-09-05 by tonyvr4