ElectronicCats/mpu6050 #20: question for the version

nices0325, 2021-02-15:
Could you let me know, what's the difference between dmp6 with dmp v6.12???

actually in the sketch, WORLD_ACCEL values are different..

On dmp6(sketch) world_accel, values are 0,0,0(x,y,z), otherwise on dmp6.12 it shows 0,0,8192(x,y,z) .. 

since I'm newbie.. i really need your help!

Thanks for your good library again!

sylvanoMTL, 2021-02-15:
Can you please provide the lines number for each code?  (and possibly copy-paste the relevant part of the code)

nices0325, 2021-02-15:
as soon as I added this code to my own sketch FIFO overflow comes out... what happend.. Actually I'm not familiar to i2c communication... so that I 'm afraid to say what's wrong

In my loop
  // if programming failed, don't try to do anything
        //if (!dmpReady) return;

        // wait for MPU interrupt or extra packet(s) available
        while (!mpuInterrupt && fifoCount < packetSize) {
            if (mpuInterrupt && fifoCount < packetSize) {
                // try to get out of the infinite loop 
                fifoCount = mpu.getFIFOCount();
            }
            // other program behavior stuff here
            // .
            // .
            // .
            // if you are really paranoid you can frequently test in between other
            // stuff to see if mpuInterrupt is true, and if so, "break;" from the
            // while() loop to immediately process the MPU data
            // .
            // .
            // .
        }

        // reset interrupt flag and get INT_STATUS byte
        mpuInterrupt = true;
        mpuIntStatus = mpu.getIntStatus();

        // get current FIFO count
        fifoCount = mpu.getFIFOCount();

        // check for overflow (this should never happen unless our code is too inefficient)
        if ((mpuIntStatus & (0x01 << MPU6050_INTERRUPT_FIFO_OFLOW_BIT)) || fifoCount >= 1024) {
            // reset so we can continue cleanly
            mpu.resetFIFO();
            fifoCount = mpu.getFIFOCount();
            Serial.println(F("FIFO overflow!"));
            
            
            FIFO overflow shows up on my serial monitor and it doesn't give me the values...

closed 2022-10-12 by stale[bot]