ElectronicCats/mpu6050 #21: Supporting devices with WOAHMI 0x98

holzachr, 2021-04-14:
There seem to be devices out there, that respond with "0x98" when asked for the WHOAMI register 0x75.
According to the datasheet, the correct answer would be "0x68". However, these "other" devices are out there, and tend to confuse the users of this library and including projects, when their material is not working.
I cannot verify if these devices work fine for the rest of the spec, but maybe someone out there can, and could extend MPU6050::testConnection() to validate the getDeviceID to be either 0x34 or 0x4C?
Does anyone have experience with such devices?

sabas1080, 2021-04-22:
These devices with different addresses from the datasheet may be Chinese clones

holzachr, 2021-04-23:
Possibly. I bought 20 MPU6050s from 3 different sellers. 7 of 20 responded with "0x68", and the other 13 responded "0x98". There's no visibile difference on the outside of the modules at all, and it appears they all come from the same production batch. It seems to be a common problem that about 50 % of the modules purchased are "clones", so why should they not be supported, if they work fine for the rest of the spec?

sabas1080, 2021-04-25:
ok, I understand, it seems fine to me, can you send a pullrequest are your code proposal?

closed 2021-09-12 by sabas1080