ElectronicCats/mpu6050 #23: Compilation of demo MPU6050_DM6 for Arduino Nano 33 BLE fails [workaround included]

robtasm, 2021-04-22:
Compilation of the example script MPU6050_DM6, freshly installed version 0.2.1, targeting the Nano 33 BLE, fails with the error:

    /Users/rob/Documents/Arduino/libraries/MPU6050/src/MPU6050.cpp: In member function 'void MPU6050::PrintActiveOffsets()':
    /Users/rob/Documents/Arduino/libraries/MPU6050/src/MPU6050.cpp:3309:154: error: 'dtostrf' was not declared in this scope

In `packages/arduino/hardware/mbed_nano/2.0.0/cores/arduino` there is `api/deprecated-avr-comp/avr/dtostrf.c.impl` which suggests adding the line

    #include "api/deprecated-avr-comp/avr/dtostrf.c.impl"

to the top of MPU6050.cpp would provide this definition. That makes compilation work for me; for a release you might have to wrap that `#include` in some `#ifdef` protection.

sabas1080, 2021-04-22:
Hi @robtasm 

Can you send pullrequest with the changes please? Thank you

sabas1080, 2021-04-25:
Solved #24

closed 2021-04-25 by sabas1080