ElectronicCats/mpu6050 #31: Library claims ARM support but uses dtostrf

renato-grottesi, 2021-09-13:
The library claims ARM support but uses dtostrf, making it incompatible with Arduino Due.
Removing the call to dtostrf, the code compiles, but accelgyro.testConnection() in he raw example fails to connect.

sylvaneau, 2021-09-18:
Hi renato,

The arduino Due is actually a SAM architecture, which is currently unsupported by this library. SAM and SAMD architecure needs to manually include `avr/dtostrf.h` in order to work properly.

However it is just a matter of few changes to make it compatible. I've just submitted a PR in order to fix it.

renato-grottesi, 2021-10-14:
@sylvaneau 
sorry for the late answer, but I fried my previous accelerometer module and I had to wait for a new one to be delivered.
Your change works well and I can now read the data from my Arduino Due!
Many thanks! :-D