ElectronicCats/mpu6050 #38: Orientation

doojinkang, 2022-05-08:
I have investigated MPU6050 library and teapot demo example in MPU6050/examples/MPU6050_DMP6.
I found dmpGetYawPitchRoll use orientation -180degree rotated from the MPU6050 hardware orientation.
Teapot demo uses toxiclibs, also roll pitch direction is diff from MPU6050 hardware orientation.

[image]
(1) MPU6050 hardware orientation by the datasheet
(2) Orientation of dmpGetYawPitchRoll by the lib
(3) Orientation in toxiclibs.

Please share any opinion.
Thank you

doojinkang, 2022-05-10:
Hello @Arthulisher
I'm not going to say the orientation is wrong or need fix.
Orientation (2) is widely accepted in aero, so called NED.
Orientation (3) is toxiclibs dependent, but the X, Y is switched in the application.
I found the teapot application X, Y is different from the MPU6050 spec, so I had some study to know why.
I hope this information is helpful.
Thank you

Arthulisher, 2022-05-10:
Hello @doojinkang
Thanks for your feedback

Yes, the information that you share is very helpful to us and our community 
We thank you for that.

best regards
ElectronicCats support team

closed 2022-06-28 by jazhe