ElectronicCats/mpu6050 #40: Will it be 0 by default according to the angle of my boot?

bbhxwl, 2022-07-17:
Will it be 0 by default according to the angle of my boot?

My device has been tilted. When it is turned on, it is still 0 for X, y and Z. how can I get x, y and Z based on the earth?. Not affected by power on and off?

[image]

bbhxwl, 2022-07-17:
```
#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include "Wire.h"
MPU6050 mpu;
uint8_t devStatus;
uint8_t mpuIntStatus;
uint16_t packetSize;
void setup() {
    Wire.begin();
    Wire.setClock(400000);
    Serial.begin(115200);
    mpu.initialize();
    if (mpu.testConnection()) {
        Serial.println("ok");
    }
    else {
        Serial.println("no");
    }
    devStatus = mpu.dmpInitialize();
    mpu.setXGyroOffset(220);
    mpu.setYGyroOffset(76);
    mpu.setZGyroOffset(-85);
    mpu.setZAccelOffset(1788);
    mpu.CalibrateAccel(6);
    mpu.CalibrateGyro(6);
    mpu.PrintActiveOffsets();
    mpu.setDMPEnabled(true);
    if (devStatus == 0) {
        Serial.println("dmp ok");
    }
    else
    {
        Serial.println("dmp no");
    }
}
uint8_t fifoBuffer[64];
Quaternion q;
VectorFloat gravity;    // [x, y, z]            gravity vector
float ypr[3];
void loop() {

    
    //获取数据
    mpu.dmpGetCurrentFIFOPacket(fifoBuffer);
        //把fifoBuffer 转换后给q
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    //获取重力    好像不太重要？
    mpu.dmpGetGravity(&gravity, &q);
    //不知道是干什么的
    mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);
    Serial.print("ypr\t");
    Serial.print(ypr[0] * 180 / M_PI);
    Serial.print("\t");
    Serial.print(ypr[1] * 180 / M_PI);
    Serial.print("\t");
    Serial.println(ypr[2] * 180 / M_PI);
}
```

Eric286, 2022-07-18:
Hello there!
Thank you for contacting ElectronicCats Support Team,
To help us to understand your situation, we would like to ask you for the following additional information:
-Which board are you using?
-What Operating System and version are you using (e.g., Windows 11, macOS 12.0, Linux)?
-Which IDE are you using to program the board?

Yes, it could be due to the angle of the boot, have you calibrated the sensor?

It shouldn't be affected if you turn it on or off.

To make it based on the earth, it could only be done with the z axis, using magnetic earth module to display Z rotation

Have a nice day!
Kind Regards!

bbhxwl, 2022-07-19:
> Hello there! Thank you for contacting ElectronicCats Support Team, To help us to understand your situation, we would like to ask you for the following additional information: -Which board are you [...]

I used the 
Arduino IDE, 
esp32, 
mpu6050 
 windows10.
Every time I start, x, y and Z are based on the current position of 0. I don't know whether mpu6050 can't do the right angle or I can't use it?

sylvanoMTL, 2022-07-19:
What happens if you use your own offset and comment the calibration routines:

mpu.CalibrateAccel(6);
mpu.CalibrateGyro(6);

?
As

Eric286, 2022-07-19:
Here the most likely situation is that your mpu6050 can't do the right angle. This due to external factors.

---------------------------------------------------------------------------------------------------------------------

" What happens if you use your own offset and comment the calibration routines:
mpu.CalibrateAccel(6);
mpu.CalibrateGyro(6); "

In this part if you comment on the calibration part, this would not be done and you would have to set the appropriate values yourself (doing test after test to see which is correct for a given position) and each time you initialize the program you would have to put the sensor in the exact same way the for the values that you provided.

Eric286, 2022-07-19:
> What happens if you use your own offset and comment the calibration routines:
> 
> mpu.CalibrateAccel(6); mpu.CalibrateGyro(6);
> 
> ?

In this part if you comment on the calibration part, this would not be done and you would have to set the appropriate values yourself (doing test after test to see which is correct for a given position) and each time you initialize the program you would have to put the sensor in the exact same way the for the values that you provided.

bbhxwl, 2022-07-20:
Is it true that 6050 does not support obtaining the true angle based on itself?

sylvanoMTL, 2022-07-20:
> Look at my video. 
The video is not available, I can only listen to some background noise. 

> Is it true that 6050 does not support obtaining the true angle based on itself?
As any 6-axis IMU sensor, raw data (acceleration and rotation velocity from the gyroscope) are processed (fusion) to obtained output angles, namely Yaw/Pitch/Roll. Referencing on a 6-IMU is made based on a calibrated reference. 

For true headings, you would need to incorporate a magnetometer to reference with earth. You could look at incorporating a magnetometer or 9-axis IMU. once again you would need to "fusion" the raw data to get the true orientation. Internet provide a large litterature on how mems IMU sensors work and how raw data can be processed.

bbhxwl, 2022-07-20:
> The video is not available, I can only listen to some background noise.
>
>
> As any 6-axis IMU sensor, raw data (acceleration and rotation velocity from the gyroscope) are processed (fusion) to [...]
Look at the video I sent.

[IMG_2667.zip](https://github.com/ElectronicCats/mpu6050/files/9150458/IMG_2667.zip)

sylvanoMTL, 2022-07-20:
The file is not acessible

bbhxwl, 2022-07-20:
> The file is not acessible

Did you see?

sylvanoMTL, 2022-07-20:
Not even accessible on my side

bbhxwl, 2022-07-21:
> Not even accessible on my side

Is this the sharing link of onedrive? It can't be opened, can it? Are you in China?

sylvanoMTL, 2022-07-22:
All of your links are not available. Passed that point, please propose something readable and thoroughly explained (as possible keeping this on github). Without any proper communication, no one would be able to help you.

As stated above your current code is making a calibration of the IMU in the setup loop. If you tilt the sensor and press reset, you are setting up a new "zero" position. To avoid this, you need to comment the calibration functions for both the accelerometer and the gyroscope, and provide your default calibration factors using the `mpu.setXGyroOffset(#youroffset#);` functions

bbhxwl, 2022-07-22:
> All of your links are not available. Passed that point, please propose something readable and thoroughly explained (as possible keeping this on github). Without any proper communication, no one [...]

Can you really do this? I use translation software, maybe I can't speak English well., Do you have email? I'll send you my video. Have a look.
I may not understand the specific use of the code at each step.

bbhxwl, 2022-07-22:
> All of your links are not available. Passed that point, please propose something readable and thoroughly explained (as possible keeping this on github). Without any proper communication, no one [...]

In the exchange group, many people told me that 6050 can't do this, but I can buy other people's modules, which are also 6050.

bbhxwl, 2022-07-22:
> mpu.setXGyroOffset(220);
>     mpu.setYGyroOffset(76);
>     mpu.setZGyroOffset(-85);
>     mpu.setZAccelOffset(1788);
>     mpu.CalibrateAccel(6);
>     mpu.CalibrateGyro(6);
>     mpu.PrintActiveOffsets();

```
mpu.setXGyroOffset(220);
    mpu.setYGyroOffset(76);
    mpu.setZGyroOffset(-85);
    mpu.setZAccelOffset(1788);
    mpu.CalibrateAccel(6);
    mpu.CalibrateGyro(6);
    mpu.PrintActiveOffsets();
```

Is my code unimportant? Can it be removed? Don't run every time?

bbhxwl, 2022-07-22:
I don't know how to write the specific code so that the value of each startup remains the same? Ask for help

```
#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps20.h"
#include "Wire.h"
MPU6050 mpu;
uint8_t devStatus;
uint8_t mpuIntStatus;
uint16_t packetSize;
void setup() {
    Wire.begin();
    Wire.setClock(400000);
    Serial.begin(115200);
    mpu.initialize();
    if (mpu.testConnection()) {
        Serial.println("ok");
    }
    else {
        Serial.println("no");
    }
    devStatus = mpu.dmpInitialize();
  /*  mpu.setXGyroOffset(0);
    mpu.setYGyroOffset(0);
    mpu.setZGyroOffset(0);
    mpu.setZAccelOffset(0);
    mpu.CalibrateAccel(6);
    mpu.CalibrateGyro(6);
    mpu.setXGyroOffset(0);*/
    mpu.PrintActiveOffsets();
    mpu.setDMPEnabled(true);
    if (devStatus == 0) {
        Serial.println("dmp ok");
    }
    else
    {
        Serial.println("dmp no");
    }
}
uint8_t fifoBuffer[64];
Quaternion q;
VectorFloat gravity;    // [x, y, z]            gravity vector
float ypr[3];
void loop() {

    //获取数据
    mpu.dmpGetCurrentFIFOPacket(fifoBuffer);
    //把fifoBuffer 转换后给q
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    //获取重力    好像不太重要？
    mpu.dmpGetGravity(&gravity, &q);
    //不知道是干什么的
    mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);
    Serial.print("ypr\t");
    Serial.print(ypr[0] * 180 / M_PI);
    Serial.print("\t");
    Serial.print(ypr[1] * 180 / M_PI);
    Serial.print("\t");
    Serial.println(ypr[2] * 180 / M_PI);
}
```

bbhxwl, 2022-07-22:
> All of your links are not available. Passed that point, please propose something readable and thoroughly explained (as possible keeping this on github). Without any proper communication, no one [...]

video  
https://pan.bbhxwl.com/s/dq8hv
https://pan.bbhxwl.com/s/D2Zcm

Eric286, 2022-07-22:
> > > Look at my video.
> > 
> > 
> > The video is not available, I can only listen to some background noise.
> > > Is it true that 6050 does not support obtaining the true angle based on itself?
> > 
> > 
> > As any 6-axis IMU sensor, raw data (acceleration and rotation velocity from the gyroscope) are processed (fusion) to obtained output angles, namely Yaw/Pitch/Roll. Referencing on a 6-IMU is made based on a calibrated reference.
> > For true headings, you would need to incorporate a magnetometer to reference with earth. You could look at incorporating a magnetometer or 9-axis IMU. once again you would need to "fusion" the raw data to get the true orientation. Internet provide a large litterature on how mems IMU sensors work and how raw data can be processed.
> 
> It seems that you can't send videos here? That is to say, I can't just rely on 6050 to get the real angle? It can only be realized with the help of other hardware, right?

As sylvanoMTL menstion earlier you would need a magnetometer to have the real angle, a gyroscope can have many factors that alter the measurements and do not give you something close to what you are looking for, it all depends on the precision you are looking for.

bbhxwl, 2022-07-23:
> As sylvanoMTL menstion earlier you would need a magnetometer to have the real angle, a gyroscope can have many factors that alter the measurements and do not give you something close to what you are [...]

Can mpu9250 get the real angle?

No related libraries found

bbhxwl, 2022-07-23:
It seems that when I test, I don't set the offset every time. Five seconds after starting, the three values of YpR, p=x and r=z, are credible. There is basically no big difference between these two values. If y=y, the difference between these values will be very large.

Can I trust the data of X and Z axes? MPU6050

sylvanoMTL, 2022-07-23:
I watched only one video and I think you have no understanding about what a MEMs Inertial measurement unit works.
There are many discussions on MEMS Inertial measurement unit. [here](http://www.starlino.com/imu_guide.html) is a good start.

In short: 

1.  raw data are computed to output yaw/pitch/roll.

    The algorithm is either developed on the microcontroller, or obtained by the DMP (on board processor) of the MPU6050 which has been reversed engineer. The algorithm you developed are more flexible (e.g. you compute euler angles, or quaternions), causing some overhead to the MCU. the DMP allow to release this overhead.
2. raw data are : 
           .   3 acclerations (AccX,AccY,AccZ): very noisy but values are around 0
           .   3 rotational speed (GyroX,GyroY,GyroZ): not much noise but values are around a specific constant. 

3. The processing (also called fusion) rely on:
           . taking a part of the Accelerometer
           . taking a part of the INTEGRATED value of the Gyroscope, hence you end up building a constant drift
                 **This is why the offset needs to be compensated with your own value**
           . Use some filters (e.g. kalman filter, compensation filter, madwick filter) to denoise the result
           . The choice of the output: Yaw/Pitch/Roll, quaternion, Euler Angle.

The provided code from the example uses: the DMP with Yaw Pitch Roll representation. 

The only thing that you would need is to know the default offset of your MPU6050. There are other piece of codes provided for this. Once you have determined them, you do not need to use the calibration methods, but you still need to provide the offsets.

For using the MPU9250, you could have a look at this [library](https://github.com/hideakitai/MPU9250) 

Finally: 

- You never can trust data obtained using a sensor built around a Physics you do not understand, and code you do not understand. 

- Also please be aware you can edit your posts to avoid flooding the issue thread.

bbhxwl, 2022-07-23:
> I watched only one video and I think you have no understanding about what a MEMs Inertial measurement unit works. There are many discussions on MEMS Inertial measurement unit. [here](http://www.starlino.com/imu_guide.html) is a good start.
> 
> In short:
> 
> 1. raw data are computed to output yaw/pitch/roll.
>    The algorithm is either developed on the microcontroller, or obtained by the DMP (on board processor) of the MPU6050 which has been reversed engineer. The algorithm you developed are more flexible (e.g. you compute euler angles, or quaternions), causing some overhead to the MCU. the DMP allow to release this overhead.
> 2. raw data are :
>    .   3 acclerations (AccX,AccY,AccZ): very noisy but values are around 0
>    .   3 rotational speed (GyroX,GyroY,GyroZ): not much noise but values are around a specific constant.
> 3. The processing (also called fusion) rely on:
>    . taking a part of the Accelerometer
>    . taking a part of the INTEGRATED value of the Gyroscope, hence you end up building a constant drift
>    **This is why the offsets need to be compensated with your own calibration values**
>    . Use some filters (e.g. kalman filter, compensation filter, madwick filter) to denoise the result
>    . The choice of the output: Yaw/Pitch/Roll, quaternion, Euler Angle.
> 
> The provided code from the example uses: the DMP with Yaw Pitch Roll representation.
> 
> The only thing that you would need is to know the default offsets of your MPU6050. There are other piece of codes provided for this. Once you have determined them, you do not need to use the calibration methods, but you still need to provide the offsets.
> 
> For using the MPU9250, you could have a look at this [library](https://github.com/hideakitai/MPU9250)
> 
> Finally:
> 
> * You never can trust data obtained using a sensor built around a Physics you do not understand, and code you do not understand.
> * Also please be aware you can edit your posts to avoid flooding the issue thread.

Is there the easiest code for novices to get the angle. I now seem to understand why I restarted from 0 before, because I set the offset every time. Seems to be getting better if I don't get the offset every time. But I still don't understand the specific principle, maybe because my English is not good, I use Google Translate. But I still want to learn, is the accuracy of 6050 very poor? What numbers can be trusted? acceleration and angular velocity? These two values are completely believable, right? The angle I currently see seems to be believable if the offset is not set every time, the Y axis and the Z axis. The data in the first 5 seconds of booting up seems to be incorrect. After a few seconds, the data of the Y and Z axes are accurate.

Eric286, 2022-07-25:
> Is there the easiest code for novices to get the angle. I now seem to understand why I restarted from 0 before, because I set the offset every time. Seems to be getting better if I don't get the [...]

The MPU-6050 is a MEMS sensor, it is noisy and drifts.
Keep in mind that the measurements that you receive may be modified by the noise that exists and that interferes with the sensor. Its biggest advantage is that The MPU6050 module is relatively easy to get up and running and capturing the raw data output of the device.

Obviously the result does not vary so much from the real one, but see the applications in which you are going to use it and if it is really necessary that you have that much accuracy.
Another advantage of the module is that its repeatability and resolution is quite good and it can be reliable.

Is the accuracy of 6050 very poor?  Nope, But it can be easily affected.
What numbers can be trusted? You can trust all the data, just take into account the degree of complexity you need, you have to start distrusting the data if this can cause someone to be hurt or something smilar.

I recommend you see the datasheet of the sensor for more information that I know can help you.

bbhxwl, 2022-07-26:
> 

I just want to know the angle of the nursing bed. This is not very sensitive to angle. I just haven't figured out that there are only 6050, can I get the real angle? What can be trusted? For example, the angle of X and y can be trusted, but Z doesn't seem to be trusted?

bbhxwl, 2022-07-26:
> Is there the easiest code for novices to get the angle. I now seem to understand why I restarted from 0 before, because I set the offset every time. Seems to be getting better if I don't get the [...]

I just want to know the angle of the nursing bed. This is not very sensitive to angle. I just haven't figured out that there are only 6050, can I get the real angle? What can be trusted? For example, the angle of X and y can be trusted, but Z doesn't seem to be trusted?

Eric286, 2022-07-26:
> I just want to know the angle of the nursing bed. This is not very sensitive to angle. I just haven't figured out that there are only 6050, can I get the real angle? What can be trusted? For [...]

The Accelerometer sends X, Y, and Z acceleration forces. We need to convert the forces into X, Y, Z 3D angle to determine the 3D Orientation of the sensor.

The MPU6050 consists of a 3-axis Gyroscope with Micro Electro Mechanical System(MEMS) technology. It is used to detect rotational velocity along the X, Y, Z axes.

These velocities are interpreted as angles and are very close to the actual angle.

--What can be trusted?
--You can rely on all three axes, X, Y and Z

bbhxwl, 2022-07-27:
> The Accelerometer sends X, Y, and Z acceleration forces. We need to convert the forces into X, Y, Z 3D angle to determine the 3D Orientation of the sensor.
>
> The MPU6050 consists of a 3-axis [...]

If I only use 6050 without gyroscope, it doesn't matter that the angle obtained is not so accurate, can it?
I just want to know the general angle of the bed.

[image]

Eric286, 2022-07-27:
> 

For this application that you are developing, it will be very reliable.

it doesn't matter that the angle obtained is not so accurate, can it?

-The variation of the real angle is not very great, it will be useful even if you use it without a giscope.

bbhxwl, 2022-07-28:
> 

Thank you, then I see. Esp32+mpu6050, about 30 yuan, 6 dollars, is very cheap.

I just need to run the calibration for the first time, and then directly read the angles x, y, Z, right? My observation seems that y is not accurate every time, and X and y are reliable every time.

bbhxwl, 2022-07-28:
> > 
> 
> For this application that you are developing, it will be very reliable.
> 
> it doesn't matter that the angle obtained is not so accurate, can it?
> 
> -The variation of the real angle is not very great, it will be useful even if you use it without a giscope.

Thank you, then I see. Esp32+mpu6050, about 30 yuan, 6 dollars, is very cheap.

I just need to run the calibration for the first time, and then directly read the angles x, y, Z, right? My observation seems that y is not accurate every time, and X and y are reliable every time.

Eric286, 2022-07-28:
> Thank you, then I see. Esp32+mpu6050, about 30 yuan, 6 dollars, is very cheap.
>
> I just need to run the calibration for the first time, and then directly read the angles x, y, Z, right? My [...]

I just need to run the calibration for the first time, and then directly read the angles x, y, Z, right?

--Yes, that is correct. n.n

Eric286, 2022-08-04:
Dear @bbhxwl 

This is to remind you that your request is still pending and awaiting your feedback.

We would like to hear back from you about your previous request in order to know if the difficulty you were facing has been already solved or we can provide further assistance.

Thank you!

Have a good day!

Eric286, 2022-08-08:
Hello @bbhxwl 

I will close this issue.
in case you face an issue, please contact us again so we help you to find the best solution.

closed 2022-08-08 by Eric286