ElectronicCats/mpu6050 #41: Is there a library to read mpu9250 angles?

bbhxwl, 2022-07-23:
Is there a library to read mpu9250 angles?

xpeqex, 2022-07-28:
Hi @bbhxwl !

Thank you for contacting Electronic Cats support team!

We set our focus on giving you the best experience and support when using Electronic Cats products. Thus, our ability to give advice and support on non-library-supported products is limited.

Since this library is not made for this module (MPU9250), I kindly recommend you to surf online to find an official library for this module.

If you have any other queries related to Electronic Cats products don't hesitate to contact us.
Have an excellent day!

bbhxwl, 2022-08-02:
> Hi @bbhxwl !
> 
> Let us know if you still need the support or if we can close this issue.

No problem.

xpeqex, 2022-08-02:
Great, I'll close the issue now.

If you have any other queries don't hesitate to contact us.

closed 2022-08-02 by xpeqex