ElectronicCats/mpu6050 #42: When I move my mpu6050, sometimes the data will stay unchanged.

bbhxwl, 2022-07-28:
When I move my mpu6050, sometimes the data will stay unchanged.

Is it because I moved somewhere? Or how? For example, when I swing 6050, its X, y, Z will not change sometimes, but it will be normal after restart. It shouldn't be that my DuPont line isn't plugged in, right?

Eric286, 2022-07-28:
> When I move my mpu6050, sometimes the data will stay unchanged.
> 
> Is it because I moved somewhere? Or how? For example, when I swing 6050, its X, y, Z will not change sometimes, but it will be normal after restart. It shouldn't be that my DuPont line isn't plugged in, right?

Maybe the sensor is not working optimally, do you have any other sensor to try with your board and see if it works correctly?

Since the 6050 sensor may be damaged.

Eric286, 2022-07-28:
> > 
> 
> If I want to set the current status to 0, do I execute the following code? Does parameter 0 mean to set the current state to 0?
> 
> mpu.setXGyroOffset(0); mpu.setYGyroOffset(0); mpu.setZGyroOffset(0); mpu.setZAccelOffset(0); mpu.CalibrateAccel(6); mpu.CalibrateGyro(6); mpu.setXGyroOffset(0);

The offset is something that serves to counterbalance or to compensate your angle.

For example if the angle gives you 60 and you measure that it should actually give 62, you would have to put an offset of 2 to compensate for this.

bbhxwl, 2022-07-28:
> For example if the angle gives you 60 and you measure that it should actually give 62, you would have to put an offset of 2 to compensate for this.

If I don't want to know what his actual angle is now, I just want to know that the sensor thinks it is 0 degrees now, and when I turn it on later, it all thinks it is 0 degrees now. Then get the angle of his rotation? Because I want to get the angle of the nursing bed, I just want the nursing bed to be 0 degrees when it is flat, and then get how much it is raised. As for whether he lays flat at 10 degrees or 20 degrees, it doesn't matter to me, I think he is 0 degrees.

Eric286, 2022-07-28:
> If I don't want to know what his actual angle is now, I just want to know that the sensor thinks it is 0 degrees now, and when I turn it on later, it all thinks it is 0 degrees now. Then get the [...]

That's what the offset would be for.
You would have to put the bed in position 0 and see what the sensor tells you, if for example it tells you that the angle is 2 degrees in the offset, you put those degrees of difference and now the sensor interprets that this would be your 0 angle.

bbhxwl, 2022-07-28:
> That's what the offset would be for. You would have to put the bed in position 0 and see what the sensor tells you, if for example it tells you that the angle is 2 degrees in the offset, you put [...]

If I lay it flat, I know that its X angle is 5, which one should I set? Can't I set the angle offset directly? Can only set acceleration offset and angular velocity offset?
CalibrateAccel  and  CalibrateGyro   and  set*GyroOffset   and set*AccelOffset 
Is there any difference between setting offset and calibration?

```
mpu.set*GyroOffset(5);
mpu.set*AccelOffset(5);
mpu.CalibrateAccel(5);
mpu.CalibrateGyro(5);
```

Eric286, 2022-07-29:
> If I lay it flat, I know that its X angle is 5, which one should I set? Can't I set the angle offset directly? Can only set acceleration offset and angular velocity offset? CalibrateAccel and [...]

if when you leave it flat it tells you that the angle is 5, your offset will be 5, so that the program interprets that there is that difference that it has to compensate.

If you can reset the offset directly as you have in the code you just gave me.

mpu.set*GyroOffset(5);
mpu.set*AccelOffset(5);

I recommend you run the program varying only these variables, so you can see how the angle varies depending on what you put inside.

Can only set acceleration offset and angular velocity offset?

-With this library yes

Is there any difference between setting offset and calibration?

-YES there is a difference. Calibration is the process of comparing the values obtained by a measuring instrument with the corresponding measurement of a reference standard. and offset means something that serves to counterbalance or to compensate for something else.

bbhxwl, 2022-07-29:
> if when you leave it flat it tells you that the angle is 5, your offset will be 5, so that the program interprets that there is that difference that it has to compensate.
>
> If you can reset the [...]

Thank you. I understand half of it, but I still don't understand the difference between calibration and offset. Sorry, my English may not be good.

Eric286, 2022-08-02:
> Thank you. I understand half of it, but I still don't understand the difference between calibration and offset. Sorry, my English may not be good.

Do not worry.

Let me explain it in a differnet way.

Offset:
It is what is used to balance things.
To balance one effect against an opposing effect, so that there is no great difference as a result.

Calibration:
Compare the values obtained by a measuring instrument with the corresponding measurement of a reference standard.

Eric286, 2022-08-03:
Do not worry.

Let me explain it in a differnet way.

Offset:
It is what is used to balance things.
To balance one effect against an opposing effect, so that there is no great difference as a result.

Calibration:
Compare the values obtained by a measuring instrument with the corresponding measurement of a reference standard.

bbhxwl, 2022-08-04:
> 

Let me experiment. Thank you for helping me.

Eric286, 2022-08-04:
@bbhxwl 
In case you face an issue or difficultt, please contact us again so we can help you to find the best solution.

Eric286, 2022-08-04:
@bbhxwl 
Sure, I will wait for your answer

bbhxwl, 2022-08-04:
> @bbhxwl Sure, I will wait for your answer

ypr，
Is ypr[0] coordinate y?
ypr[0] =y  ? 
ypr[1]=x  ? 
ypr[2]=z  ? 

```
void loop() {

    reconnectWifi();
    reconnectMqtt();
    client.loop();
    //获取数据
    mpu.dmpGetCurrentFIFOPacket(fifoBuffer);
    mpu.dmpGetQuaternion(&q, fifoBuffer);
    mpu.dmpGetGravity(&gravity, &q);
    //不知道是干什么的
    mpu.dmpGetYawPitchRoll(ypr, &q, &gravity);
    String c = "{\"angleY\":\"";
    c.concat(ypr[0] * 180 / M_PI);
    c.concat("\"");
    c.concat(",\"angleX\":\"");
    c.concat(ypr[1] * 180 / M_PI);
    c.concat("\"");
    c.concat(",\"angleZ\":\"");
    c.concat(ypr[2] * 180 / M_PI);
    c.concat("\"}");
    Serial.println(c.c_str());
    client.publish("jiaoDuShangBao", c.c_str());
    delay(100);

}
```

Eric286, 2022-08-04:
@bbhxwl  

YPR stands for Yaw Pitch Roll. are the axes of the plane.

Pitch, yaw and roll are the three dimensions of movement when an object moves through a medium.

The terms may be used to describe an aeroplane´s movements through the air. They are also applied to watercraft moving through water, and spacecraft moving through space.

There are in fact six degrees of freedom of a rigid body moving in three-dimensional space.

As the movement along each of the three axes is independent of each other and independent of the rotation about any of these axes, the motion has six degrees of freedom (see diagram).

-pitch
nose up or tail up.
-yaw
nose moves from side to side
-roll
a circular (clockwise or anticlockwise) movement of the body as it moves forward

bbhxwl, 2022-08-05:
> @bbhxwl
>
> YPR stands for Yaw Pitch Roll. are the axes of the plane.
>
> Pitch, yaw and roll are the three dimensions of movement when an object moves through a medium.
>
> The terms may be used to describe [...]

[image]

> pitch

Why is this in the positive direction? I can tell which way to turn is YpR

Eric286, 2022-08-05:
@bbhxwl 

The positive directions would be like the ones seen in the following image.

[image]

Eric286, 2022-08-08:
Dear @bbhxwl 

This is to remind you that your request is still pending and awaiting your feedback.

We would like to hear back from you about your previous request in order to know if the difficulty you were facing has been already solved or we can provide further assistance.

Thank you!

Have a good day!

Eric286, 2022-08-12:
Hello @bbhxwl!

I will close this issue.
In case you face an issue, please contact us again so we help you to find the best solution.

closed 2022-08-12 by Eric286