ElectronicCats/mpu6050 #43: ElectronicCats

Ganishk, 2022-08-06:
I haven't able to find the proper documentation for this library. All of them were either examples or online tutorials, which explains the same code again and again. Could you please provide us with a documentation?

Eric286, 2022-08-08:
Hello @Ganishk!

What kind of documentation are you looking for?
I await your response to see it with the engineering team and see what we have to provide you.

Best regards!
Eric Chavez, Support Team

Eric286, 2022-08-15:
Dear @Ganishk 

This is to remind you that your request is still pending and awaiting your feedback.

We would like to hear back from you about your previous request in order to know if the difficulty you were facing has been already solved or we can provide further assistance.

Thank you!

Have a good day!

Eric286, 2022-08-16:
Hi @Ganishk 

The documentation that we have is the one that is already commented in each file of this library, in addition to all the files that we have in the src folder that are also already commented.
For example if we see the I2Cdev.cpp file in the src folder, we can see how it tells us what the lines of code like the following do.

```
/** Read a single bit from an 8-bit device register.
  * @param devAddr I2C slave device address
  * @param regAddr Register regAddr to read from
  * @param bitNum Bit position to read (0-7)
  * @param data Container for single bit value
  * @param timeout Optional read timeout in milliseconds (0 to disable, leave off to use default class value in I2Cdev::readTimeout)
  * @return Status of read operation (true = success)
  */
int8_t I2Cdev::readBit(uint8_t devAddr, uint8_t regAddr, uint8_t bitNum, uint8_t *data, uint16_t timeout) {
     uint8_t b;
     uint8_t count = readByte(devAddr, regAddr, &b, timeout);
     *data = b & (1 << bitNum);
     returncount;
}

```

Where even the return and the parameters are already mentioned.

If you have any questions with certain lines of code or about any use of these, you could tell me which ones specifically and thus help you in a better way.

Ganishk, 2022-08-17:
> Hi @Ganishk
>
> The documentation that we have is the one that is already commented in each file of this library, in addition to all the files that we have in the src folder that are also already [...]

Yeah thanks, now only I noticed these. But it would be easier if it was also available in a PDF or HTML or README.md file. Because when I asked, this question, I was in a hurry to finish it. So, I didn't have the patience to go through the source code. But now I found it thanks again. Also, many might be like me. Hence, I'm requesting for a **PDF file as documentation**. Anyways, I know that it might not be easy to provide now. I hope you'll provide us soon.

Eric286, 2022-08-17:
Dear @Ganishk

Currently, we do not have a date for the documentation output in this format: pdf and readme.md.
However, we are already working on it, and possibly the following week there will be an estimated date for this.

Most likely it will be uploaded or displayed in the readme.

Can we close this issue now?

Have an amazing day!

Kind regards!
Support Team.

closed 2022-08-19 by Ganishk