ElectronicCats/mpu6050 #45: ElectronicCats usage of teapot demo rotation system

bumboo001, 2022-08-23:
Hello,
I am making hand controller for VR using your library.I am using unity and I don't know how to use same system of rotating as on teapot demo.

    // toxiclibs direct angle/axis rotation from quaternion (NO gimbal lock!)

    // (axis order [1, 3, 2] and inversion [-1, +1, +1] is a consequence of

    // different coordinate system orientation assumptions between Processing

    // and InvenSense DMP)

    float[] axis = quat.toAxisAngle();

Currently i'm just sending yaw pitch roll date over bluetooth but it acts weird. how can I use same system as in teapot demo?

If further information is needed I can send them

thanks

xpeqex, 2022-08-23:
Hello @bumboo001 !
Thank you for contacting us,
Please let us know the following:

- What board are you using?

- What Operating System and version are you using (e.g., Windows 11, macOS 12.0, Linux)?

- Which IDE are you using to program the board?

Kind regards.

bumboo001, 2022-08-23:
> Hello @bumboo001 ! Thank you for contacting us, Please let us know the following:
> 
>     * What board are you using?
> 
>     * What Operating System and version are you using (e.g., Windows 11, macOS 12.0, Linux)?
> 
>     * Which IDE are you using to program the board?
> 
> 
> Kind regards.

I am using arduino nano and sometimes arduino uno. I have latest windows 10 and I'm using arduino IDE `https://www.arduino.cc/en/software`

Thanks

xpeqex, 2022-08-24:
@bumboo001

xpeqex, 2022-08-24:
@bumboo001 
Can you share with us a screenshot of the issue you are facing?
Can you indicate to me the exact name of the example you are using?

bumboo001, 2022-08-24:
@xpeqex 
> @bumboo001 Can you share with us a screenshot of the issue you are facing? Can you indicate to me the exact name of the example you are using?

At first I have issue with `[MPU6050_DMP6.ino](https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino)` because when I define `#define OUTPUT_READABLE_YAWPITCHROLL` There will be weird drift after I rotate my gyroscope maybe 90 degrees left(roll). This drift is extremely large and more than 100 degrees.(tomorrow i can make video to show it if necessary) I think that this problem is caused by gimbal lock. Sorry I am new at this.

And my solution for problem is I want to use same system as in teapot demo. Teapot demo works 100% OK. So I want to use it for my VR. The only problem is that I don't understand  what `output that matches the
// format used for the InvenSense teapot demo` means. Is it just a quaternion?

So I just need explanation **how teapot demo gets rotation** from gyroscope because comments don't explain enough for me.

bumboo001, 2022-08-30:
> @bumboo001 I've been testing the library with several MPU6050 modules and the data I read is accurate, so probably you are having some noise in the signal or there is a problem with your module.
> 
> About the teapot demo, since this project was not developed by Electronic Cats we have limited knowledge of the teapot project, however, as you may notice there is a line on the code that says
> 
> ```
>  // uncomment "OUTPUT_TEAPOT" if you want output that matches the
> // format used for the InvenSense teapot demo
> ```
> 
> so please uncomment line 120 : `#define OUTPUT_TEAPOT`, by deleting the two slashes **//**, and comment line 103: `#define OUTPUT_READABLE_YAWPITCHWOLL` by adding two slashes **//** in front of it. This should work.
> 
> Lastly, answering your question about what does it mean by InvenSense format? , this format as you mentioned is quaternion values, you can find this information as well in the example in line 328.
> 
> Hope you find this helpful. Kind regards.

Thanks for answer. This is what for searching for.

Regards

closed 2022-08-30 by bumboo001