ElectronicCats/mpu6050 #46: Which code should I use to set the current location to 0

bbhxwl, 2022-08-28:
Which code should I use to set the current location to 0

I want to get the X angle. For example, my x angle is 10 °. How can I return to 0 with the current angle

xpeqex, 2022-08-29:
Hi @bbhxwl 
Thank you for contacting us!

If you want to set an specific angle as **_0_** you have to change the corresponding axe offset in your code, in the example you propose you have to set an offset of 10 degrees so when you put the sensor in a 10 degrees what you will read will be 0.

Only modifying the offset code you will get the behave you want.

Have an excellent day!
Kind regard.

bbhxwl, 2022-08-30:
> Hi @bbhxwl Thank you for contacting us!
>
> If you want to set an specific angle as **_0_** you have to change the corresponding axe offset in your code, in the example you propose you have to set an [...]

Which of these two sets the angle X, y, Z?

```
mpu.setXGyroOffset(0);
  mpu.setYGyroOffset(0);
  mpu.setZGyroOffset(0);
```

```
mpu.setXAccelOffset(1788)
mpu.setYAccelOffset(1788)
mpu.setZAccelOffset(1788)
```

bumboo001, 2022-08-30:
@bbhxwl 
> Which of these two sets the angle X, y, Z?
>
> ```
> mpu.setXGyroOffset(0);
>   mpu.setYGyroOffset(0);
>   mpu.setZGyroOffset(0);
> ```
>
> ```
> mpu.setXAccelOffset(1788)
> mpu.setYAccelOffset(1788) [...]

This is rotation offset:
`mpu.setXGyroOffset(0);
  mpu.setYGyroOffset(0);
  mpu.setZGyroOffset(0);`

[image]

bbhxwl, 2022-08-30:
> 

Join me, gyrox is 10, gyroy is 20, gyroz is 30, I need to set
```
mpu.setXGyroOffset(10); mpu.setYGyroOffset(20); mpu.setZGyroOffset(30);

```
Is this still the case?
```
mpu.setXGyroOffset(0); mpu.setYGyroOffset(0); mpu.setZGyroOffset(0);

```

xpeqex, 2022-08-30:
@bbhxwl 
What you need to do is to first keep your module steady with the angle and position you want to start, read the values of the serial monitor, and then in your code write the value you obtain (in your case GyroX and GyroZ) in the serial monitor to modify the offset so you can have an 0 when is in a different position.
If you don't mind me asking the following, are you using our library or the Arduino official one?

Kind regards,

bbhxwl, 2022-08-31:
> setYGyroOffset

I use your library. Only your library has an angle. I may not understand what you said. I mean, if my current x angle is 10, I want it to become 0. Should I set offset setXGyroOffset(0)? setXGyroOffset(10)?  setXGyroOffset(-10)?

bbhxwl, 2022-08-31:
> @bbhxwl What you need to do is to first keep your module steady with the angle and position you want to start, read the values of the serial monitor, and then in your code write the value you obtain [...]

What value should I set to correct him

xpeqex, 2022-08-31:
@bbhxwl 
In order to set an offset please follow these instructions:

1. Comment the lines where the module calibrates by adding //
2. In lines  ```
mpu.setXGyroOffset(1400);
    mpu.setYGyroOffset(1400);
    mpu.setZGyroOffset(-38); //
    mpu.setZAccelOffset(1688); // 1688 factory default for my test chip
```
modify the values in whatever axes you need them.
3. Let the module stabilize, the module usually needs some time to get steady in the lecture.
4. Calibrate the offsets depending on your findings.

Notes
1. I use the MPU6050_DMP6 example for this
2. mpu.CalibrateAccel() and mpu.CalibrateGyro() may be more useful, you only have to put the module in the initial angle you want to be 0 and upload or reset the board.
3. This process is a calibration process so there is no specific value you have to put to obtain the exact value you are looking for, you will have to try it on your own.
Kind regards.

xpeqex, 2022-08-31:
@bbhxwl
In order to set an offset please follow these instructions:

Comment the lines where the module calibrates by adding // in front of the text
In lines 
```
mpu.setXGyroOffset(1400);
mpu.setYGyroOffset(1400);
mpu.setZGyroOffset(-38); //
mpu.setZAccelOffset(1688); // 1688 factory default for my test chip
```
modify the values in whatever axes you need them.
3. Let the module stabilize, the module usually needs some time to get steady in the lecture.
4. Calibrate the offsets depending on your findings.

Notes
1. I use the MPU6050_DMP6 example for this
2. mpu.CalibrateAccel() and mpu.CalibrateGyro() may be more useful, you only have to put the module in the initial angle you want to be 0 and upload or reset the board.
3. This process is a calibration process so there is no specific value you have to put to obtain the exact value you are looking for, you will have to try it on your own.
Kind regards.

bbhxwl, 2022-09-01:
> @bbhxwl In order to set an offset please follow these instructions:
>
> Comment the lines where the module calibrates by adding // in front of the text In lines
>
> ```
> mpu.setXGyroOffset(1400); [...]

mpu.CalibrateAccel() 和 mpu.CalibrateGyro() 

Just these two sentences, can we make the current value 0?

xpeqex, 2022-09-01:
@bbhxwl 
Yes, you need to comment on these two lines, so in this way, the module won't calibrate itself when starting.
```
//mpu.CalibrateAccel();
//mpu.CalibrateGyro();
```
Unfortunately, I am not able to give you a specific offset value to set your module's angle from 10 to 0, as I mentioned you will need to test it on your own and try with different values since none of the modules are equal in calibration even if these are the same model.

Kind regards,

bbhxwl, 2022-09-02:
> @bbhxwl Yes, you need to comment on these two lines, so in this way, the module won't calibrate itself when starting.
>
> ```
> //mpu.CalibrateAccel();
> //mpu.CalibrateGyro();
> ```
>
> Unfortunately, I am not [...]

I'm a little confused. Can't these two lines of code automatically set the current position to 0?

bbhxwl, 2022-09-02:
> @bbhxwl Yes, you need to comment on these two lines, so in this way, the module won't calibrate itself when starting.
>
> ```
> //mpu.CalibrateAccel();
> //mpu.CalibrateGyro();
> ```
>
> Unfortunately, I am not [...]

I tried to calibrate, but after the device restarts, he still needs to recalibrate?

xpeqex, 2022-09-02:
> I'm a little confused. Can't these two lines of code automatically set the current position to 0?

Yes, if you left the code as it is, with these two lines the module will calibrate itself each time the code is uploaded or reset, taking the current position as 0, meaning that if you start your module in a 10 degrees position the code will show you 0. 

But if you comment on these two lines you have to set your own offset values to calibrate the module so when the module is in a 10-degree position the lecture will be 0.

bbhxwl, 2022-09-02:
> ```
> mpu.setXGyroOffset(0);
>   mpu.setYGyroOffset(0);
>   mpu.setZGyroOffset(0);
> ```
> 
> 
>     
>   
> 
> ```
> mpu.setXAccelOffset(1788)
> mpu.setYAccelOffset(1788)
> mpu.setZAccelOffset(1788)
> ```

mpu.setXGyroOffset(0);
  mpu.setYGyroOffset(0);
  mpu.setZGyroOffset(0);
mpu.setXAccelOffset(1788)
mpu.setYAccelOffset(1788)
mpu.setZAccelOffset(1788)

I record the offset myself. Can I set this every time I turn on? But how do I get these six current values for this value? I can only get the angle and acceleration, not the angular velocity. How to get these six values? Can you send me the code?

(I may be stupid, sorry, I can read the code, but my English is too poor.)

bumboo001, 2022-09-02:
@bbhxwl 
is this what you were searching for?

https://youtu.be/sWkJ6iukjQk

bbhxwl, 2022-09-03:
> @bbhxwl is this what you were searching for?
> 
> https://youtu.be/sWkJ6iukjQk

Can you remember your offset when you turn on this?

bumboo001, 2022-09-03:
@bbhxwl 
upload this code to arduino and test if its Ok
```
// I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v2.0)
// 6/21/2012 by Jeff Rowberg <jeff@rowberg.net>
// Updates should (hopefully) always be available at https://github.com/jrowberg/i2cdevlib
//
// Changelog:
//      2019-07-08 - Added Auto Calibration and offset generator
//		   - and altered FIFO retrieval sequence to avoid using blocking code
//      2016-04-18 - Eliminated a potential infinite loop
//      2013-05-08 - added seamless Fastwire support
//                 - added note about gyro calibration
//      2012-06-21 - added note about Arduino 1.0.1 + Leonardo compatibility error
//      2012-06-20 - improved FIFO overflow handling and simplified read process
//      2012-06-19 - completely rearranged DMP initialization code and simplification
//      2012-06-13 - pull gyro and accel data from FIFO packet instead of reading directly
//      2012-06-09 - fix broken FIFO read sequence and change interrupt detection to RISING
//      2012-06-05 - add gravity-compensated initial reference frame acceleration output
//                 - add 3D math helper file to DMP6 example sketch
//                 - add Euler output and Yaw/Pitch/Roll output formats
//      2012-06-04 - remove accel offset clearing for better results (thanks Sungon Lee)
//      2012-06-01 - fixed gyro sensitivity to be 2000 deg/sec instead of 250
//      2012-05-30 - basic DMP initialization working

/* ============================================
I2Cdev device library code is placed under the MIT license
Copyright (c) 2012 Jeff Rowberg

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in
all copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
THE SOFTWARE.
===============================================
*/

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu;
//MPU6050 mpu(0x69); // <-- use for AD0 high

/* =========================================================================
[... 262 lines cut ...]
            Serial.print(aaWorld.y);
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}
```

bbhxwl, 2022-09-03:
> 

OK, I will reply to you later after the test

bbhxwl, 2022-09-05:
> @bbhxwl upload this code to arduino and test if its Ok
>
> ```
> // I2C device class (I2Cdev) demonstration Arduino sketch for MPU6050 class using DMP (MotionApps v2.0)
> // 6/21/2012 by Jeff Rowberg [...]

Every time I initialize, I need to remember the X, y, Z of YpR, save it, and add this value to + every time, right? Ensure that the last memorized value can be read even after power on

bumboo001, 2022-09-05:
@bbhxwl yes you understand code correctly.
If you want you can save offsets to eeprom so you will be able to reed theese values after restarting arduino.

xpeqex, 2022-09-06:
@bbhxwl You can find that information documented in the code, in line 143 you can find the declaration of ypr array:
`float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector`

xpeqex, 2022-09-12:
Hello @bbhxwl !
This is a message to remind you that your request has been pending for 5 days and awaits your comments. 

I would like to hear from you about your previous request to see if the difficulty you were facing has been resolved. Otherwise, if we do not hear back from you within the next three days, the ticket will be closed.

Kind regards,
Electronic Cats Support Team

closed 2022-09-13 by bbhxwl