ElectronicCats/mpu6050 #50: Library version should be bumped up.

shaise, 2022-09-09:
Hi, 
Thanks for your library!
I think the library version should be bumped up to 0.5.1, here is why:
I just added your library to my ESP32 PlatformIO project, but it did not work.
After several hours of debugging I found out there is an extra unneeded `Wire.beginTransmission(...);` before `Wire.requestFrom(...)`.
I wanted to report it as bug, but I found out it was already fixed here. However since the lib version was not changed, you still get the non working version when you install the lib with PlatformIO.
***Thanks a lot***

xpeqex, 2022-09-09:
Hello @shaise!
Thank you for contacting Electronic Cats Support Team. 

Truly appreciated you have taken the time out of your day to comment on this, I will communicate this to my colleagues to do a new release including what you mentioned. Feel free to get back to us if you have any further queries.

Would you like to be informed?

Have an amazing day!
Kind regards!
Support Team.

sabas1080, 2022-09-09:
Ready https://github.com/ElectronicCats/mpu6050/releases/tag/v0.6.0

closed 2022-09-09 by sabas1080

shaise, 2022-09-10:
***Thanks for the fast response!***