ElectronicCats/mpu6050 #51: esp32 2.0.4 Unlimited waiting for bugs

bbhxwl, 2022-09-19:
esp32 2.0.4 Unlimited waiting for bugs
```
    mpu.initialize();

```
When the program runs here, it will not run the following code. Arduino for esp32 1.0.6 is normal.

Eric286, 2022-09-19:
Dear @bbhxwl !
Thank you for contacting us.
To better understand your situation, please, we require the following information:

Could you explain your problem a little more? Since from what I understand the code does not compile you in that line of code.
Can I have the error message?
Have you had this problem before? If so, when?
Has anything changed since this issue started?
Do you experience this issue on another computer\device\network?
What IDE are you using and what version of it?

In the same way, we require you to send us any photograph or video that you think is necessary to better understand your problem.
Have a nice day!
Kind regards!
Support Team.

bbhxwl, 2022-09-19:
> Dear @bbhxwl ! Thank you for contacting us. To better understand your situation, please, we require the following information:
>
> Could you explain your problem a little more? Since from what I [...]

I use vscode+plaformIO 2.0.4

The code can only run before this sentence, and will not run after this sentence, and there is no error prompt.

Eric286, 2022-09-19:
Hello @bbhxwl !

This may be due to the fact that the library was not installed correctly and it does not know how to interpret that line.
Could you please try the program in the Arduino IDE?
Also, with any of the examples does this error happen to you?
Also please try to check the example MPU6050_raw.ino and tell me if it works for you.

bbhxwl, 2022-09-20:
> Hello @bbhxwl !
>
> This may be due to the fact that the library was not installed correctly and it does not know how to interpret that line. Could you please try the program in the Arduino IDE? Also, [...]

[image]
The same, infinite waiting.

bbhxwl, 2022-09-20:
> Hello @bbhxwl !
>
> This may be due to the fact that the library was not installed correctly and it does not know how to interpret that line. Could you please try the program in the Arduino IDE? Also, [...]

Arduino must be OK. I always use this.

bbhxwl, 2022-09-20:
> Hello @bbhxwl !
>
> This may be due to the fact that the library was not installed correctly and it does not know how to interpret that line. Could you please try the program in the Arduino IDE? Also, [...]

[image]
Can it be repaired?

bbhxwl, 2022-09-20:
> Hello @bbhxwl !
>
> This may be due to the fact that the library was not installed correctly and it does not know how to interpret that line. Could you please try the program in the Arduino IDE? Also, [...]

[image]

Eric286, 2022-09-20:
Hello @bbhxwl !

Sorry for my late response.

"Can it be repaired?"

-Please give me some time to test it and see, if necessary, how to fix it or what should be done to fix it.

Eric286, 2022-09-21:
Dear @bbhxwl !

I have already done the tests and this is what I found.
All lines of code worked for me, however, the way I can replicate your error was by disconnecting the voltage from the sensor, with this it always stays on the same line as you.
A very big possibility is that your module is damaged since the code worked perfectly with the same characteristics that you sent me in the last messages.

bbhxwl, 2022-09-21:
> Dear @bbhxwl !
>
> I have already done the tests and this is what I found. All lines of code worked for me, however, the way I can replicate your error was by disconnecting the voltage from the sensor, [...]

It can't be broken. I can use arduino ide esp32 1.0.6 to run normally.

Eric286, 2022-09-21:
Hello @bbhxwl !

Could you check if you don't have some other libraries called MPU6050 that may be interfering? since in all the tests that I have done, it does not happen to me like you, seeing that it is not a problem of the library.

Do you have the Arduino extension in VSCode?

bbhxwl, 2022-09-22:
> Hello @bbhxwl !
> 
> Could you check if you don't have some other libraries called MPU6050 that may be interfering? since in all the tests that I have done, it does not happen to me like you, seeing that it is not a problem of the library.
> 
> Do you have the Arduino extension in VSCode?

plaformIO
[image]

Eric286, 2022-09-22:
Hello @bbhxwl !

Can you please install this extension and follow the instructions I will give you to solve the problem?

[image]

After installing this extension please upload the sketch you like for example I uploaded the example called MPU6050_DMP6.ino

[image]

You can do all this in the bottom bar of VScode, make sure that the board is correct as well as the port.

[image]

[image]

At the top, the symbol must appear to be able to compile the program.

[image]

And then you should be able to see the console. And it shouldn't stay on the mpu.initialize();

Eric286, 2022-09-27:
Hello @bbhxwl!

Is your issue solved?

Best regards!
Support Team.

bbhxwl, 2022-09-28:
> Hello @bbhxwl!
> 
> Is your issue solved?
> 
> Best regards! Support Team.

No, I still use the 1 version temporarily, instead of the 2 version.

Eric286, 2022-09-28:
Hello @bbhxwl !

Could you tell me what you mean by the first version or what you mean by this?

Eric286, 2022-10-03:
Hello @bbhxwl !

Can we close the issue?

Eric286, 2022-10-06:
Hello @bbhxwl!

We will now close this issue.

In case you have any problems or comments, do not hesitate to contact us again.

Have an amazing day!

Kind regards!
Support Team.

closed 2022-10-06 by Eric286