ElectronicCats/mpu6050 #54: Wrong reading when the MPU6050 is placed vertically

Juno-cyber, 2022-12-06:
I tried to initialize the mpu6050 when it was **placed vertically**, but the mpu6050 got wrong data. The error is that when I rotate the mpu6050, the angle change cannot correspond to my deflection. **Is there a way to set the initialization to use the mpu6050 vertically?**

I have been stuck with this problem for a long time. Thanks for anybody who can give me some advise.

xpeqex, 2022-12-07:
Hi @Juno-cyber 

Thank you for contacting us!

Most of the examples included in this repository have an auto calibrations routine that runs at the beginning of the code. Meaning that if you place your MPU6050 module in the initial position you need and run the code the values will be 0. 

These are the examples that run this mentioned routine: 

- [IMU_Zero](https://github.com/ElectronicCats/mpu6050/tree/master/examples/IMU_Zero)
- [MPU6050_DMP6](https://github.com/ElectronicCats/mpu6050/tree/master/examples/MPU6050_DMP6)
- [MPU6050_DMP6_using_DMP_V6v12](https://github.com/ElectronicCats/mpu6050/tree/master/examples/MPU6050_DMP6_using_DMP_V6v12)

However, let me inform you that you may find some variations in the values because each module varies.

Lastly, you can always modify the code, changing the calibrations times, turning off the calibrations, and changing the offset values to get more accurate results.

Hope you find this information helpful.
Have a nice day!

xpeqex, 2022-12-14:
Hi @Juno-cyber 
Let us know if there is anything else we can assist you with or if we can consider this issue closed.

PTT31, 2023-08-13:
hi
Stm32f401rct6 and mpu6050
i was used imu_zero examples for set offset(Stm32f401rct6 and mpu6050), but it didn't stop after 30 minutes 
[image]
help me!!

xpeqex, 2023-08-14:
Hello @PTT31 

It would be best if you opened a [new GitHub issue](https://github.com/ElectronicCats/mpu6050/issues) to track this issue. This allows us to track the issue more easily and make sure that it gets the attention it deserves.

Here is some information we request you to include in your issue:

- What board are you using.
- What IDE are you using?
- What version of the library are you using?
- The code that you are using.
- The setup that you have.
- A description of the issue that you are facing.
- Any steps that you have taken to try to fix the issue.