ElectronicCats/mpu6050 #55: Reading problem when placing the mpu6050 vertically

Juno-cyber, 2023-01-31:
### **Transplant example MPU6050_ DMP6 encountered a problem. The angle measurement is incorrect after vertical initialization**

When I try automatic calibration of acceleration and attitude angle, initial three-axis attitude angle after calibration is about 0.When I place the mpu6050 horizontally (the z+direction is downward), the three-axis angle changes correctly.
But when I place the mpu6050 vertically (the x+direction is downward), the measured values of the three-axis angle will change correspondingly, but the measured results tend to decay. For example, if I rotate the pitch from 0 ° to 60 °, the measured results will increase from 0 ° to 60 °, but cannot reach 60 °, it will decay all the time。

Is there some suggestions for improvement in this situation？

***Thanks a lot***

AndreaZGuz, 2023-01-31:
Dear @Juno-cyber!
Thank you for contacting us.

Most of the examples included in this repository have an auto calibrations routine that runs at the beginning of the code. Meaning that if you place your MPU6050 module in the initial position you need and run the code the values will be 0.
If you are using this example [MPU6050_DMP6](https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino) from line 165 to line 246 you will see the calibration process.
You can try modifying the offset values between parentheses on lines: 206, 207, 208, and 209; and maybe also modify the calibration times on lines 214 and 215.

However, let me inform you that you may find some variations in the values because each module varies.

I hope this information was helpful to you!
Have a nice day
Kind regards,
Electronic Cats Support Team.

AndreaZGuz, 2023-02-01:
Dear @Juno-cyber!

Thank you for the feedback.
I would like to share with you the outcomes I had testing the library [MPU6050_DMP6](https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino) using a MPU6050 module I have for testing.
As you will see, the values reached -66 and more, up to -74. 

ypr	71.19	-59.48	7.39
ypr	71.43	-59.81	7.30
ypr	71.67	-60.12	7.21
ypr	71.90	-60.42	7.12
ypr	72.14	-60.72	7.05
ypr	72.38	-61.03	6.97
ypr	72.63	-61.36	6.90
ypr	72.90	-61.72	6.82
ypr	73.17	-62.09	6.78
ypr	73.45	-62.46	6.75
ypr	73.69	-62.80	6.73
ypr	73.93	-63.10	6.72
ypr	74.15	-63.38	6.69
ypr	74.36	-63.64	6.66
ypr	74.58	-63.91	6.62
ypr	74.82	-64.22	6.58
ypr	75.06	-64.53	6.52
ypr	75.30	-64.85	6.47
ypr	75.53	-65.16	6.44
ypr	75.75	-65.44	6.40
ypr	75.95	-65.71	6.37
ypr	76.16	-65.98	6.34
ypr	76.36	-66.26	6.32
ypr	76.57	-66.55	6.30
ypr	76.77	-66.83	6.29
ypr	76.97	-67.11	6.28
ypr	77.16	-67.38	6.27
ypr	76.36	-66.26	6.32
ypr	76.57	-66.55	6.30
ypr	76.77	-66.83	6.29
ypr	76.97	-67.11	6.28
ypr	77.16	-67.38	6.27
ypr	77.34	-67.62	6.25
ypr	77.53	-67.86	6.20
ypr	77.69	-68.09	6.17
ypr	77.86	-68.30	6.11
ypr	78.01	-68.48	6.03
ypr	78.15	-68.66	5.97
ypr	78.30	-68.85	5.88
ypr	78.45	-69.05	5.81
ypr	78.61	-69.25	5.69
ypr	78.76	-69.44	5.55
ypr	78.88	-69.61	5.44
ypr	79.02	-69.77	5.30
ypr	79.15	-69.93	5.17
ypr	79.53	-70.43	4.84
ypr	79.64	-70.57	4.77
ypr	79.73	-70.71	4.73
ypr	79.82	-70.84	4.69
ypr	79.93	-70.97	4.61
ypr	80.02	-71.10	4.53
ypr	80.11	-71.22	4.46
ypr	80.20	-71.35	4.38
ypr	80.31	-71.48	4.31
ypr	80.42	-71.63	4.22
ypr	80.53	-71.79	4.15
ypr	80.63	-71.94	4.09
ypr	80.73	-72.10	4.05
ypr	80.83	-72.25	4.03
ypr	81.65	-73.54	4.04
ypr	81.65	-73.57	4.09
ypr	81.66	-73.61	4.15
ypr	81.66	-73.63	4.22
ypr	81.65	-73.65	4.33
ypr	81.64	-73.67	4.44
ypr	81.63	-73.69	4.54
ypr	81.62	-73.71	4.65
ypr	81.62	-73.74	4.74
ypr	81.62	-73.76	4.78
ypr	81.63	-73.78	4.80
ypr	81.64	-73.80	4.81
ypr	81.64	-73.80	4.81
ypr	81.63	-73.79	4.81
ypr	81.63	-73.79	4.81
ypr	81.63	-73.78	4.82
ypr	81.63	-73.78	4.85
ypr	81.70	-73.97	5.12
ypr	81.70	-73.98	5.13
ypr	81.71	-74.01	5.14
ypr	81.71	-74.01	5.18
ypr	81.70	-74.02	5.24
ypr	81.69	-74.02	5.31
ypr	81.69	-74.03	5.36
ypr	81.68	-74.03	5.41
ypr	81.68	-74.04	5.41
ypr	81.69	-74.05	5.42
ypr	81.68	-74.06	5.44
ypr	81.68	-74.06	5.44
ypr	81.69	-74.07	5.46
ypr	81.69	-74.09	5.49
ypr	81.69	-74.10	5.50
ypr	81.69	-74.10	5.53
ypr	81.68	-74.11	5.56

If the calibration section on your code is executed successfully and you are experimenting with issues just with the pitch, probably there is an error on the hardware.
Please, think about testing with another MPU5060 module. It might help.

It is ok If we close this issue?
Have a nice day!
Regards!

Juno-cyber, 2023-02-03:
Dear @AndreaZGuz, Thanks for all of your help!

I'm going to make sure if there some wrong with the hardware.

Sincerely thank you again!
May you have a nice day!
Juno-cyber.

closed 2023-02-03 by Juno-cyber