ElectronicCats/mpu6050 #56: i get stuck in  Success! DMP code written and verified

IamEden94, 2023-02-02:
02:33:36.853 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
02:33:37.253 -> *wm:AutoConnect: SUCCESS 
02:33:37.253 -> *wm:STA IP Address: 192.168.0.11
02:33:37.253 -> WiFi connected! IP address: 192.168.0.11
02:33:37.253 -> Initializing I2C devices...
02:33:37.253 -> Testing device connections...
02:33:37.253 -> MPU6050 connection successful
02:33:37.253 -> Initializing DMP...
02:33:37.253 -> 
02:33:37.253 -> 
02:33:37.253 -> Resetting MPU6050...
02:33:37.287 -> Checking hardware revision...
02:33:37.287 -> Revision @ user[16][6] = 165
02:33:37.287 -> Resetting memory bank selection to 0...
02:33:37.287 -> Reading OTP bank valid flag...
02:33:37.287 -> OTP bank is valid!
02:33:37.287 -> Setting slave 0 address to 0x7F...
02:33:37.287 -> Disabling I2C Master mode...
02:33:37.287 -> Setting slave 0 address to 0x68 (self)...
02:33:37.287 -> Resetting I2C Master control...
02:33:37.334 -> Setting clock source to Z Gyro...
02:33:37.334 -> Setting DMP and FIFO_OFLOW interrupts enabled...
02:33:37.334 -> Setting sample rate to 200Hz...
02:33:37.334 -> Setting external frame sync to TEMP_OUT_L[0]...
02:33:37.334 -> Setting DLPF bandwidth to 42Hz...
02:33:37.334 -> Setting gyro sensitivity to +/- 2000 deg/sec...
02:33:37.334 -> Writing DMP code to MPU memory banks (1929 bytes)
02:33:37.506 -> Success! DMP code written and verified.

How to solve it?
im using wemos d1 mini

Eric286, 2023-02-02:
Dear @IamEden94 

Let me run tests to see if something might be causing this behavior.

Only, could you send me the version of the library that you are using, if it is the latest version or an old one?

And what version of the MPU6050 model are you using?

Best Regards!

IamEden94, 2023-02-03:
Hi @Eric286 

I used all latest and get this
esp8266 3.1.1
arduino ide 2.0.4
mpu6050 v0.6.0
wifimanager v0.16
osc 3.5.7

im using ITG MPU gy-521 and wemos d1 mini with breadboard

D1=SCL
D2=SDA
GND=GND
5V=VCC

01:36:29.119 -> *WM: Connecting as wifi client...
01:36:29.119 -> *WM: Status:
01:36:29.119 -> *WM: 0
01:36:29.119 -> *WM: Using last saved values, should be faster
01:36:32.435 -> *WM: Connection result: 
01:36:32.435 -> *WM: 3
01:36:32.435 -> *WM: IP Address:
01:36:32.435 -> *WM: 192.168.0.11
01:36:32.435 -> WiFi connected! IP address: 192.168.0.11
01:36:32.435 -> Initializing I2C devices...
01:36:32.435 -> Testing device connections...
01:36:32.435 -> MPU6050 connection successful
01:36:32.435 -> Initializing DMP...
01:36:32.667 -> Enabling DMP...
01:36:32.667 -> Enabling interrupt detection (Arduino external interrupt 0)...
01:36:32.667 -> DMP ready! Waiting for first interrupt...
01:36:32.667 -> *WM: freeing allocated params!

Eric286, 2023-02-03:
Hello @IamEden94 

Could you change this line please, it would only be changing 15 to 13, and now the interrupt pin would go to pin D7.

#define INTERRUPT_PIN 15 // use pin 15 on ESP8266

It is line 139 of the code.

IamEden94, 2023-02-04:
HI @Eric286 

Changed but The result is same

Eric286, 2023-02-07:
Dear @IamEden94 

1. Check your wiring and make sure all of your connections are secure.
2. Check that you have the correct board selected in your Arduino IDE. 
3. Make sure you have the latest version of the ESP8266 board libraries installed. 
4. Try resetting the device and running the code again.

Eric286, 2023-02-14:
Hello @IamEden94 

We will now close this issue.

At any time you can open it again.

In case you have any problems or comments, do not hesitate to contact us again.

Have an amazing day!

Kind regards!
Support Team.

closed 2023-02-14 by Eric286