ElectronicCats/mpu6050 #60: MPu6050 register map revision dont have all registers

linxcow, 2023-07-20:
Your library is based on register map rev 2.0 but i cant find it.

The later revisions dont have all registers.

example : Register 0x1F ( 31 ) is not  mentioned in the register map documents i found. ( setting motion detect threshold )

I know production is stopped but still a lot of GY521 are sold these days.

Eric286, 2023-07-21:
Hello @linxcow!

Thank you for bringing this to our attention, we truly appreciate your interest in the library.

We assure you that we take your request seriously, and we will thoroughly investigate the differences and work towards upgrading our library to provide the most accurate and up-to-date information possible.

If there are any other questions or clarifications you seek, please don't hesitate to reach out to us. We are here to help you in any way we can.

Best regards,

Support Team!

Eric286, 2023-07-21:
Hello @linxcow!

If you want to find more information about the library, please go to the following repository, from which we rely, for more information.

We do not have the document you require.

https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050

Best Regards!

closed 2023-07-22 by linxcow