ElectronicCats/mpu6050 #64: Support for GIGA R1 WiFi with 3 x I2C

megacct, 2023-11-09:
Hi, been using your library successfully on a Mega (thankyou) but I'm now migrating to a GIGA and would like to use Wire1 rather than Wire. I notice other mpu6050 libs have the interface as an argument. Is that possible currently (maybe with a local hack) with your lib? If not, any plans to do this as I'd like to stick with this lib.

Thanks, Steve

xpeqex, 2023-12-01:
Hi @megacct !

Sorry for the late reply. We really appreciate your support in this library, we appreciate you brining this to us.

At the moment there is no a "simple" way to set the library to use Wire1 instead of Wire as you mentioned. Nevertheless, there is a local hack and I just tested it out.

Here are the changes I made to make it work with the GIGA.

- Locate the relevant files:

1. Go to the library folder. This is usually located at C:\Users\<username>\Documents\Arduino\libraries.
2. Locate the Electronic Cats MPU6050 library folder. This is usually named MPU6050.
3. Open the `I2Cdev.cpp` file in your preferred text editor. I recommend you Visual Studio code to perfom the following steps more effectively.

- Modify the code

1. Search for instances where Wire is used directly.
2. Replace those instances with a variable that can hold either Wire or Wire1 based on the chosen argument, or you can directly replace it Wire with Wire1. What I did is to use the find and replace function of Visual Studio Code to first find all the `Wire.` definitions and the change them to `Wire1.` this way I avoided changing the library Wire definition.
3. Replace the instances Wire to Wire1 in the Arduino sketch you are using.

- Test with Arduino GIGA.
I have tested this with the DMP6 example and the GIGA is able to communicate with the sensor and work with it. I will share this feedback with my colleagues since it would be great to find an easier way to implement this.

Hope it works for you. Let us know if there are any other questions.

xpeqex, 2023-12-01:
@megacct Oh nice. Well I assume you did it before my reply 😅. I will share this feedback with my colleagues to work on this and include a better way to add the GIGA (Wire1) compatibility to the library.

Please feel free to contact us again in case you have any other queries.

closed 2023-12-01 by xpeqex