ElectronicCats/mpu6050 #65: Changing sensors in code on lsm303dlh and l3gd20

brightproject, 2023-11-12:
Hello @sabas1080 
I want to use gyroscope and accelerometer sensors like `lsm303dlh` and `l3gd20`.
How to find the “entry point” to insert your own acceleration and angular velocity values into the code?
I get the raw data something like this:
```
#include <Wire.h>
#include <L3G.h>
#include <LSM303.h>

LSM303 mag_accel;
L3G gyro;

void setup() {
  Serial.begin(115200);
  Wire.begin();
  if (!gyro.init())
  {
    Serial.println("Failed to autodetect gyro type!");
    while (1);
  }
  if (!mag_accel.init())
  {
    Serial.println("Failed to initialize mag_accel!");
    while (1);
  }
  gyro.enableDefault();
  mag_accel.enableDefault(); 
}

void loop() {

  gyro.read();
  mag_accel.read();

Serial.print(gyro.g.x);
Serial.print(",");
Serial.print(gyro.g.y);
Serial.print(",");
Serial.print(gyro.g.z);
Serial.print(",");
Serial.print(mag_accel.a.x);
Serial.print(",");
Serial.print(mag_accel.a.y);
Serial.print(",");
Serial.println(mag_accel.a.z);

}
```
As far as I understand, your code examples contain source code for working with [DMP](https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino) and [more](https://github.com/ElectronicCats/mpu6050/blob/master/examples/IMU_Zero/IMU_Zero.ino)?

When I saw a dump of some code in the [code](https://github.com/ElectronicCats/mpu6050/blob/3779bf6d4f0bbebb35b4de88c00129ae3a7598ec/src/MPU6050_6Axis_MotionApps20.h#L140), it seemed to me that your code was not applied to other sensors other than MPU9250.
I would be grateful for any hint.

xpeqex, 2023-11-21:
Hi @brightproject 

Thank you for contacting us!

As you stated, this library is developed for its use exclusively with MPU6050, which is why you will only find definitions and files referring to the MPU6050. 

---------------------------
> How to find the “entry point” to insert your own acceleration and angular velocity values into the code?

I am not sure what you are meaning by this question, could you please rephrase it for me?

---------------------------

> As far as I understand, your code examples contain source code for working with [DMP](https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino) and [more](https://github.com/ElectronicCats/mpu6050/blob/master/examples/IMU_Zero/IMU_Zero.ino)?

Correct. DMP (Digital Motion Processor) is an algorithm for the raw data to convert it into Euler but this feature is exclusive to the MPU6050 and MPU9250.  The MPU-6050 devices combine a 3-axis gyroscope and a 3-axis accelerometer on the same silicon die, together with an onboard Digital Motion Processor™ (DMP™), which processes complex 6-axis MotionFusion algorithms. Since this processor is a hardware component it is not possible to adapt or integrate other sensors that do not include this DMP.

brightproject, 2023-11-23:
> I am not sure what you are meaning by this question, could you please rephrase it for me?

Thank you for your response.
I call the entry point the section of code where the raw data of the accelerometer and angular velocities are inserted into the code.
In the example code
https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino
as you already noted, all the work is done by the exclusive DMP algorithm.
Therefore, there is no way to find a place where raw data on accelerations and angular velocities is obtained.
I was interested in this comment
https://github.com/ElectronicCats/mpu6050/blob/3779bf6d4f0bbebb35b4de88c00129ae3a7598ec/examples/MPU6050_DMP6/MPU6050_DMP6.ino#L16
Because I am looking for the possibility of compensating for non-gravitational forces using `9DOF.`
But apparently without DMP it is impossible to use your code, for example with other sensors.

xpeqex, 2023-11-23:
@brightproject 

Gravity information is obtained from the on-chip DMP and a variable resistor on the board that measures the gravity of the environment using a PID control in the [MPU6050.cpp file](https://github.com/ElectronicCats/mpu6050/blob/master/src/MPU6050.cpp).

We have an example code to read and print the [raw data from the MPU6050.](https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_raw/MPU6050_raw.ino)

Again, take in mind that this library is purely developed for the MPU6050 module sensor. We recommend you use a different library for the sensors you are using, there are multiple libraries out there for different sensors.

xpeqex, 2023-11-28:
Hi here,

We will close this ticket now. I was glad to help you out answering to your questions.

If you have any other questions or concerns, reach out to us again!

closed 2023-11-28 by xpeqex