ElectronicCats/mpu6050 #66: ElectronicCats

Bojarin-Jr, 2023-11-16:
Hi, there.

I was trying to understand how to set frequency for accelerometer and gyroscope for MPU6050 using your libraries.
Could you help me guys?

Bojarin,
Thanks in advance.

xpeqex, 2023-11-17:
Hello @Bojarin-Jr 

We appreciate you bringing to us this problem. I would like to better understand the issue you are facing. So I request you provide me a further explanation of what you are working on or what you are trying to achieve.  Additionally to this please post a copy of the code you are using.

Lastly, and if I am not wrong you are referring to setting up the sample rate, correct me if not. If that is the case you can use setRate() wich is explained here: https://github.com/ElectronicCats/mpu6050/blob/3779bf6d4f0bbebb35b4de88c00129ae3a7598ec/src/MPU6050.cpp#L134-L136

In any case, I am looking forward to your reply!

Bojarin-Jr, 2023-11-20:
Hi there)

I am working on accelerometer data processing. My goal is to make MPU 6050 to measure acceleration at least with a frequency of about 500-600 Hz. I found out your library and would like to adapt it to my task.

xpeqex, 2023-11-23:
@Bojarin-Jr 

It is possible to achieve a sampling rate of 500-600 Hz for accelerometer data using the library. However, it's important to note that the actual achievable sampling rate may vary depending on the specific board and the complexity of the data processing tasks being performed.

The sensor register output, FIFO output, DMP sampling, Motion detection, Zero Motion detection, and Free Fall detection are all based on the Sample Rate. 

The Sample Rate is generated by dividing the gyroscope output rate by SMPLRT_DIV.

**Sample Rate**

$$ Sample Rate = {Gyroscope Output Rate  \over (1 + \text{SMPLRT DIV}) } $$

where Gyroscope Output Rate = 8kHz when the DLPF is disabled (DLPF_CFG = 0 or 7), and 1kHz when the DLPF is enabled.

SMPLRT_DIV is a 8-bit unsigned value. 

Then you can use the function https://github.com/ElectronicCats/mpu6050/blob/3779bf6d4f0bbebb35b4de88c00129ae3a7598ec/src/MPU6050.cpp#L134-L136

You can have a complete look to this here: https://github.com/ElectronicCats/mpu6050/blob/3779bf6d4f0bbebb35b4de88c00129ae3a7598ec/src/MPU6050.cpp#L102-L136

Hope you find this helpful.

xpeqex, 2023-11-28:
Hi here,

We will close this ticket now. I was glad to help you out answering to your questions.

If you have any other questions or concerns, reach out to us again!

closed 2023-11-28 by xpeqex