ElectronicCats/mpu6050 #73: What's the command to: Open the processing sketch (Teapot).

mw66, 2024-01-16:


mw66, 2024-01-16:
found answer here:

https://forum.arduino.cc/t/mpu6050-electronic-cats-library-example-does-not-compile/1143806/7

BTW, the `~/Processing/libraries/` name on Linux is: `~/sketchbook/libraries/`.

mw66, 2024-01-16:
However for 

step 8: Run the processing sketch.

I got error: MPUTeapot.pde:77:0:77:0: RuntimeException: Error opening serial port /dev/ttyUSB0: Port busy

xpeqex, 2024-01-16:
Hi @mw66 

Thank you for sharing this feedback. These comments are really helpful for a better guidance in the use of the library and its examples. I will add these in the Wiki.

Feel free to open a new issue if you have any other suggestions!

 
<p align=right>
[image]
</p>

closed 2024-01-16 by xpeqex