ElectronicCats/mpu6050 #74: ElectronicCats

mw66, 2024-01-16:
https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6/MPU6050_DMP6.ino#L292

Is this conversion correct?

Why when I put my chip vertical (90 degree), the print output is: 28.07, i.e. (~= 90 / 3.14 = 28.66)? BTW, I got ~+/- 0.xx at zero degrees.

For example for pitch: the chip in flat position (~0 degree):

```
ypr	8.53	-0.10	0.08
ypr	8.53	-0.11	0.10
ypr	8.53	-0.12	0.10
ypr	8.53	-0.13	0.10
ypr	8.53	-0.13	0.10
ypr	8.53	-0.13	0.09
ypr	8.53	-0.13	0.09
ypr	8.53	-0.14	0.09
ypr	8.53	-0.14	0.09
ypr	8.53	-0.14	0.09
ypr	8.53	-0.14	0.10
ypr	8.53	-0.14	0.10
ypr	8.53	-0.14	0.10
```

the chip in vertical position (~90 degree):
```
ypr	12.09	28.00	14.33
ypr	12.09	28.05	14.36
ypr	12.08	28.10	14.39
ypr	12.08	28.14	14.42
ypr	12.06	28.20	14.45
ypr	12.06	28.24	14.48
ypr	12.05	28.31	14.51
ypr	12.04	28.36	14.53
ypr	12.03	28.42	14.56
ypr	12.02	28.48	14.59
ypr	12.02	28.54	14.63
ypr	12.01	28.59	14.67
ypr	12.00	28.64	14.71
ypr	12.00	28.69	14.75
ypr	12.00	28.75	14.79
ypr	11.99	28.81	14.84
ypr	11.98	28.86	14.88
ypr	11.97	28.92	14.92
```

what I'm missing?

Thanks.

mw66, 2024-01-16:
same issue also reported here:

https://github.com/jrowberg/i2cdevlib/issues/763

mw66, 2024-01-16:
@xpeqex 

Thanks for the info.

Regarding the degrees print-out, can I conclude that my chip is faulty? since you got ~90, I got ~28.

As to the hard coded offset, I found in the code:

https://github.com/ElectronicCats/mpu6050/blob/df034d8a2d1eb862c189ce27d0abfd93e6649d57/examples/MPU6050_DMP6/MPU6050_DMP6.ino#L206-L215

Shouldn't line 206 ~ 209 be all removed?

Since on line 214 ~ 215 (when the device successfully init-ed) the calibration routines will overwrite these values anyway.

If the device failed init (e.g. wrong dev address 0x68 vs. 0x69), then line 206 ~ 209 shouldn't be called at all.

xpeqex, 2024-01-16:
@mw66 

It may indicate that your module it is not working correctly, very common on affordable modules. You can try increasing the calibration time and do not move the sensor while initialize. 

Regarding the offsets and/or calibration codelines, I agree it can be commented to let the user the opportunity to use offset or the auto calibration, instead or deleting the whole codelines.

About the other suggestions, you are right, if the sensor is not initialized it is not needed to run these codelines.

jrowberg, 2024-01-17:
I'm commenting here instead of on the I2Cdevlib repo since I think the conversation does better here, in one spot. You could also be running into the issue of gimbal lock, i.e. what happens when you try to convert a 4-dimensional quaternion into 3-dimensional representation. This is particularly visible when you rotate a full 90 degrees. If you see believable numbers when the IMU is flat, with believable values reported when angles are, for example, below 60 degrees, then your IMU is probably fine.

The closer you get to 90 degrees, the larger the effect that dimensional reduction has. I expect that at a full 90 degree rotation (where Z is now perpendicular to gravity), the values reported for other axes will also vary wildly even with small movements. If this is what you observe, it's definitely gimbal lock. The problem is that there is no non-arbitrary way to represent a quaternion in 3-dimensional space. The order in which you compute each axis's rotation matters, and choosing yaw-pitch-roll (for example) will yield different angle values than roll-pitch-yaw, because there is more than one set of rotations from 0/0/0 that can achieve any orientation represented by a quaternion.

Maybe this isn't the issue, but I suspect that it is.

mw66, 2024-01-17:
Thanks for the comments.

> If you see believable numbers when the IMU is flat, with believable values reported when angles are, for example, below 60 degrees, then your IMU is probably fine.

No, when the chip move from 0 degree to 90 degree, the reported value moves from 0 => 28.xx (the biggest value), almost monotonically.

And at 45 degree (estimate), the value is about 25 ~ 26.

Also as @xpeqex said he got 90 at 90 degrees, I suspect it's more because of a faulty chip.

mw66, 2024-01-17:
@jrowberg @xpeqex Just want to let you know that I tried another library:

https://github.com/gabriel-milan/TinyMPU6050

where I can get stable ~80 degree reading for 90 degree chip position, so I think there maybe a bug in i2cdevlib (and mpu6050 since it uses the former).

xpeqex, 2024-01-17:
@mw66 

What is commented by @jrowberg is really important to highlight on this issue. Just for reference I have shared my test with the MPU6050 I have with me (sorry for the mess).  

https://github.com/ElectronicCats/mpu6050/assets/107638696/eeeb41e1-4d0e-432a-99a5-bd248c6c604b

I am writing down all the suggestions to edit our wiki and improve the sketches. 

I am adding this at the same time I received your last message, be sure we will take a look at it, is it possible for you to let us know what version, example you are using?

mw66, 2024-01-17:
I'm using version 1.2.0, example mpu6050/examples/MPU6050_DMP6/MPU6050_DMP6.ino

Yes, the video shows it works on your end, but not on mine, it's hard for you to debug.

If you can provide a debug version, I can run and send you the debug output.

xpeqex, 2024-01-18:
@mw66 
As far as I could see in the source code of the [**gabriel-milan/TinyMPU6050**](https://github.com/gabriel-milan/TinyMPU6050) repository, there is no sign of DMP use, when ours is using DMP. 

Actually they are using the sensor's raw data. So it could be the reason your module is running fine with that library, you may want to try the MPU6050_raw examples included in our library.

mw66, 2024-01-18:
@xpeqex 

https://github.com/ElectronicCats/mpu6050/blob/df034d8a2d1eb862c189ce27d0abfd93e6649d57/examples/MPU6050_raw/MPU6050_raw.ino#L131-L137

does not print the angle, which is what I need.

Can we add the angle calculation into the MPU6050_raw example?

xpeqex, 2024-01-19:
@mw66 
I see and at the moment there is no function to get the angles directly from the raw data since all this data pass through the DMP, and then it is printed/used in the sketch.  

I would be a nice feature to create an example to get the module's angles by calculating them from the raw data, I will discuss this with my colleagues, but I cannot guarantee this implementation.

jrowberg, 2024-01-19:
> I would be a nice feature to create an example to get the module's angles by calculating them from the raw data, I will discuss this with my colleagues, but I cannot guarantee this implementation.

The problem is that you can only calculate angles from raw data if you know that the IMU is not moving, and therefore all acceleration is due to gravity only. Otherwise, you can get very wrong angles if there is any other linear motion involved.

If you want an example of what's possible with only raw data (no DMP) using this or any similar 6-axis IMU, check out this video I made about two years ago:

https://twitter.com/jrowberg/status/1608662546705448960

Source is available here:

https://github.com/perilib/perilib-arduino-core/blob/master/examples/Register_TWI_MPU6050_direct_madgwick/Register_TWI_MPU6050_direct_madgwick.ino

It's built on top of the Perilib I/O library, but that's not necessary. The important part is that you regularly feed in the raw accel/gyro X/Y/Z values into the Madgwick 6-DOF fusion function.

xpeqex, 2024-01-22:
@jrowberg Thank you for sharing this and helping us out in this one, really helpful.

@mw66 Hope you find useful the code Jeff is sharing to us!

Feel free to contact us again.

closed 2024-01-22 by xpeqex