ElectronicCats/mpu6050 #76: Gyroscope

Navaneeth272001, 2024-01-24:
**Please, before reporting any issue**
- Make sure your board it's in good condition.
- Make sure your MPU6050 it's in good condition.
- Verify that it really is a library problem and not a hardware problem.
- **Avoid** to submit a GitHub issue for project troubleshooting.

Any feedback/suggestions should be discussed on the [feedback section](https://github.com/ElectronicCats/mpu6050/issues):
  * Just click on New Issue and select the correct category.

When reporting any issue, please try to provide all relevant information to help on its resolution.

**Describe the bug**
A clear and concise description of what the bug is.
The more detailed this is, the easier we can come up with a solution.

**To Reproduce**
Complete source code which can be used to reproduce the issue. Please try to be as generic as possible.
No extra code, extra hardware, etc.
If you think it is absolutely necessary to add this, mention in detail the connections in case it is necessary hardware and also indicate the part of code that was added and how it was added.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**To help us to understand your situation, we would like to ask you for the following additional information::**
 - Which board are you using?
 - What Operating System and version are you using (e.g. Windows 11, macOS 12.0, Linux)?
 - In case you are using the Arduino IDE, What version of the Arduino IDE?
 - Did it work before?, If it worked before, what were the parameters that you modify?

**Additional context**
Add any other context about the problem here that you think will help us to solve your problem.

xpeqex, 2024-01-24:
Hi @Navaneeth272001 

Please share with me the following, so I can better understand the issue:
- Version of the library you are using.
- What board are you using.
- What code are you using.
- If possible a picture of the setup you have. 

Regards,

Navaneeth272001, 2024-01-25:
Hi

Version of the library - 1.3.0
[image]

Board - MPU6050 with ESP32

Code - I am using the code MPU6050_raw but called the functions    accelgyro.CalibrateGyro(); and
  accelgyro.CalibrateAccel(); in the setup. 
  
The  accelgyro.CalibrateAccel(); function is working as expected. But  accelgyro.CalibrateGyro(); function seems to be not working. That is, if the MPU6050 module is positioned differently and started, the gyroscope values does not go to zero with respect to that position.

[image]

xpeqex, 2024-01-25:
@Navaneeth272001 

It is hard for me to try to replicate this without a complete code. Could you please share with me? 

Is using the RAW example you may want to disable the DMP to avoid interference with your code.

Navaneeth272001, 2024-01-26:
> If using the RAW example you may want to disable the DMP to avoid

First of all I would like to understand how gyroscope calibration works.
Please let me know what happens when the gyroscope calibration function is
called?

xpeqex, 2024-01-26:
Sure @Navaneeth272001 
You can check the complete routine of the calibration functions here:

https://github.com/ElectronicCats/mpu6050/blob/d81945514fc9c527403179072a5faf029054edab/src/MPU6050.cpp#L3360-L3453

Basically:

- Uses PID control to adjust gyroscope offsets.
- Accelerometer calibration uses different PID gains and addresses for accelerometer offsets.

- Looping Through Calibration Cycles:

  - Iterates through the specified number of Loops.
    - **Within each loop**:
      - Performs 100 PI calculations.
      - Reads sensor data repeatedly.
      - Calculates error and updates PID terms (proportional and integral).
      - Computes new calibration offsets.
      - Writes offsets back to sensor registers.
      - Checks for convergence and adjusts PID gains if necessary.

    - **Finalization**:
      - Resets the FIFO (First-In-First-Out) buffer and DMP (Digital Motion Processor) on the MPU6050.

Navaneeth272001, 2024-01-30:
Thanks for the details. I understood the code flow, but I would like to understand further into what is the impact of "CalibrateGyro" function on the outputs "gx, gy and gz". How will the outputs "gx, gy and gz" differ when the "CalibrateGyro" is called and when it is not called. Will there be any difference in the output? if so, under what condition the outputs will vary when the function is called and when it is not called. 

I am not able see any difference between the outputs "gx, gy and gz", when the function is called and when it is not called. To understand the work of the function, I would like to do an experiment to see the changes caused due to the calibration routine in the output when the "CalibrateGyro" is called and when it is not called. Please let me know how can I do it.

Thanks

xpeqex, 2024-01-30:
@Navaneeth272001 

Setting the offset instantly has an effect on the readings. When the MPU6050 is level and stable the calibration samples the sensors and shifts the offsets to get Zeros on 5 of the 6 axis and adjusts Accell Z to show 1g.

Our ability to give advice and support questions related to general project-based queries is limited. Additionally, since this repository is a fork to isolate the MPU6050 usage, we recommend you post it on the original repository https://github.com/jrowberg/i2cdevlib since queries like this can be answered by many people and different solutions can be explored. Also, this will help the community to grow, which will ultimately benefit users.

Navaneeth272001, 2024-01-31:
Thanks for your time and answers!

closed 2024-01-31 by Navaneeth272001