ElectronicCats/mpu6050 #80: The DEBUG build no longer compiles

benpeart, 2024-03-01:
If you attempt to compile with the DEBUG macro defined, the library does not compile with the error:

```
.pio\libdeps\debug\MPU6050\src\MPU6050_9Axis_MotionApps41.cpp: In member function 'uint8_t MPU6050_9Axis_MotionApps41::dmpInitialize()':
.pio\libdeps\debug\MPU6050\src\MPU6050_9Axis_MotionApps41.cpp:366:17: error: 'productID' was not declared in this scope
     DEBUG_PRINT(productID);
```
I don't see a way for me to submit a pull request so I have attached a patch that contains a two-part fix. The first part is to fix the compile error by commenting out the offending DEBUG_PRINT line. The second/better fix is to update the DEBUG_PRINT macros so that the structure of the macros ensures that the code is always validated at compile time but is not called when DEBUG is undefined. This will prevent stale debug code from accumulating when the debug version is not regularly compiled.

[0001-Update-DEBUG_PRINT-macros-so-that-the-code-is-always.patch](https://github.com/ElectronicCats/mpu6050/files/14463080/0001-Update-DEBUG_PRINT-macros-so-that-the-code-is-always.patch)

benpeart, 2024-03-01:
I was using MPU6050_6Axis_MotionApps20.h but any example that tries to build with the DEBUG flag set will have this error. If you look the cause is trivial - someone commented out the line in MPU6050_9Axis_MotionApps41.cpp that declares the productID variable but forgot to comment out the DEBUG_PRINT line that uses it.

I've submitted pull request #81 that fixes this with both a simple fix (comment out the line that tries to use productID) and a more robust fix that should prevent these types of errors in the future.

AndreaZGuz, 2024-03-01:
Thank you @benpeart for the comments.

Currently, the .ino examples do not have issues during compilation when DEBUG is defined, but you have a point regarding the productID definition.
Your pull request will be checked.

Is there another issue we can help you with?

closed 2024-03-07 by AndreaZGuz