ElectronicCats/mpu6050 #85: MPU6050 connection failed but showing DMP results!

Sinamajidi, 2024-08-20:
Hey there, I'm using ESP32 and I wanted to connect it to my mpu6050 chip of  mpu-92.65 board using this library.

after scanning i2c addresses i figured out its set to 0x68 and I also connected the SDA and SCL to 21, and 22 respectively.

After running the example although it couldn't connect to MPU6050 but showed me the DMP results in real-time, how it can be possible to cant be connected to MPU6050 but show the result of the DMP?! (DMP  is dependent on the MPU6050 chip) 

I know this is a bug but I dont know how to solve it. help me please.
Here is the modified example of the library to that gave me that result:

```

// I2Cdev and MPU6050 must be installed as libraries, or else the .cpp/.h files
// for both classes must be in the include path of your project
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu(0x68);
//MPU6050 mpu(0x69); // <-- use for AD0 high

#define OUTPUT_READABLE_YAWPITCHROLL

#define INTERRUPT_PIN 2  // use pin 2 on Arduino Uno & most boards
#define LED_PIN LED_BUILTIN // (Arduino is 13, Teensy is 11, Teensy++ is 6)
bool blinkState = false;

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

// packet structure for InvenSense teapot demo
uint8_t teapotPacket[14] = { '$', 0x02, 0,0, 0,0, 0,0, 0,0, 0x00, 0x00, '\r', '\n' };

// ================================================================
// ===               INTERRUPT DETECTION ROUTINE                ===
// ================================================================

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================

[... 184 lines cut ...]

```
Initializing I2C devices...
Testing device connections...
MPU6050 connection failed

Send any character to begin DMP programming and demo: 
Initializing DMP...
>**......>......3144.00000,	-3156.00000,	13110.00000,	-94.00000,	-90.00000,	-171.00000

Enabling DMP...
Enabling interrupt detection (Arduino external interrupt 2)...
DMP ready! Waiting for first interrupt...

ypr	0.00	-0.01	0.44
ypr	0.00	-0.01	0.44
ypr	0.00	-0.01	0.44
ypr	0.00	-0.01	0.44
ypr	0.00	-0.01	0.43
ypr	0.00	-0.01	0.43
ypr	0.00	-0.01	0.43

```

@sabas1080 @REGIOIGER @quesswho @JahazielLem @ignacioacosta

Eric286, 2024-08-20:
Hello @Sinamajidi 

The fact that you're seeing DMP output even after the connection failure is peculiar because the DMP relies on the MPU6050 to function.

The library might be reporting a failure incorrectly, or there could be an edge case where the DMP still processes data despite a failed connection.

Check that the power supply to the MPU6050 is stable and at the correct voltage level.

Ensure that you are using the latest version of the library.

Likewise, I will see it with my team to investigate and correct it.

Have a nice Day!

Support Team

Sinamajidi, 2024-08-20:
Great. I checked them all, and I will be thankful if you correct this bug. Before that, I'm gonna ignore that error, and I would try to use DMP independently.

sabas1080, 2024-08-20:
@Sinamajidi Hi, unfortunately since the creation of the library mpu.initialize(); does not have a return value and this makes it impossible to detect if the library is initialized correctly. It would be a question to see if devStatus = mpu.dmpInitialize(); can help you to stop the program if it is not initialized correctly. We will try to improve mpu.initialize(); so that at startup the device can do a startup validation and not continue to advance, but this will take a few months for our current projects, contributions are welcome. Thanks

Sinamajidi, 2024-08-21:
> @Sinamajidi Hi, unfortunately since the creation of the library mpu.initialize(); does not have a return value and this makes it impossible to detect if the library is initialized correctly. It would be a question to see if devStatus = mpu.dmpInitialize(); can help you to stop the program if it is not initialized correctly. We will try to improve mpu.initialize(); so that at startup the device can do a startup validation and not continue to advance, but this will take a few months for our current projects, contributions are welcome. Thanks
> 
> Edit: mpu.testConnection() may be the temporary function that helps you.
> if(mpu.testConnection() == False){
>   Serial.print("MPU6050 connection failed");
>   while(True); 
> }
> with this the program will stop if the sensor is not detected and will not continue running.

@sabas1080 
I think there is no need to do that thing you just said, bcz after any physical movement the serial monitor shows the change in those three factors of DMP simultaneously and bcz of that i think there is a problem between DMP and MPU in library source file rather than checking for connections.

sabas1080, 2024-08-22:
@Sinamajidi can you try to test this sketch and see how it behaves?
```

#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu(0x68);
//MPU6050 mpu(0x69); // <-- use for AD0 high

#define OUTPUT_READABLE_YAWPITCHROLL

#define INTERRUPT_PIN 2  // use pin 2 on Arduino Uno & most boards
#define LED_PIN LED_BUILTIN // (Arduino is 13, Teensy is 11, Teensy++ is 6)
bool blinkState = false;

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

// packet structure for InvenSense teapot demo
uint8_t teapotPacket[14] = { '$', 0x02, 0,0, 0,0, 0,0, 0,0, 0x00, 0x00, '\r', '\n' };

// ================================================================
// ===               INTERRUPT DETECTION ROUTINE                ===
// ================================================================

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================

void setup() {
    // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.setPins(21, 22);
       Wire.begin();
        Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif

[... 151 lines cut ...]
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}

```
please share with me the output of the serial terminal

Sinamajidi, 2024-08-23:
Ok, I did. The first time I saw the monitor, it was infinitely showing the following message:
```

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
ets Jul 29 2019 12:21:46

```

After that, I held the EN button on ESP32 and disconnected. After immediately reconnecting the VCC pin of the MPU module, I released the EN button and this appeared in the serial monitor:
```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Initializing I2C devices...
[    17][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[    27][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[    34][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
[    41][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
Testing device connections...
[    49][E][Wire.cpp:499] requestFrom(): i2cWriteReadNonStop returned Error -1
MPU6050 connection failed

```

sabas1080, 2024-08-26:
@Sinamajidi  Ok, this information is useful to me, can you try to test this sketch, please share with me the output of the serial terminal

```
#include "I2Cdev.h"

#include "MPU6050_6Axis_MotionApps20.h"
//#include "MPU6050.h" // not necessary if using MotionApps include file

// Arduino Wire library is required if I2Cdev I2CDEV_ARDUINO_WIRE implementation
// is used in I2Cdev.h
#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
    #include "Wire.h"
#endif

// class default I2C address is 0x68
// specific I2C addresses may be passed as a parameter here
// AD0 low = 0x68 (default for SparkFun breakout and InvenSense evaluation board)
// AD0 high = 0x69
MPU6050 mpu(0x68);
//MPU6050 mpu(0x69); // <-- use for AD0 high

#define OUTPUT_READABLE_YAWPITCHROLL

#define INTERRUPT_PIN 2  // use pin 2 on Arduino Uno & most boards
#define LED_PIN LED_BUILTIN // (Arduino is 13, Teensy is 11, Teensy++ is 6)
bool blinkState = false;

// MPU control/status vars
bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer

// orientation/motion vars
Quaternion q;           // [w, x, y, z]         quaternion container
VectorInt16 aa;         // [x, y, z]            accel sensor measurements
VectorInt16 aaReal;     // [x, y, z]            gravity-free accel sensor measurements
VectorInt16 aaWorld;    // [x, y, z]            world-frame accel sensor measurements
VectorFloat gravity;    // [x, y, z]            gravity vector
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

// packet structure for InvenSense teapot demo
uint8_t teapotPacket[14] = { '$', 0x02, 0,0, 0,0, 0,0, 0,0, 0x00, 0x00, '\r', '\n' };

// ================================================================
// ===               INTERRUPT DETECTION ROUTINE                ===
// ================================================================

volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high
void dmpDataReady() {
    mpuInterrupt = true;
}

// ================================================================
// ===                      INITIAL SETUP                       ===
// ================================================================

void setup() {
    // join I2C bus (I2Cdev library doesn't do this automatically)
    #if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
        Wire.setPins(21, 22);
       Wire.begin();
        Wire.setClock(400000); // 400kHz I2C clock. Comment this line if having compilation difficulties
    #elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
        Fastwire::setup(400, true);
    #endif

[... 151 lines cut ...]
            Serial.print(aaWorld.y);
            Serial.print("\t");
            Serial.println(aaWorld.z);
        #endif
    
        #ifdef OUTPUT_TEAPOT
            // display quaternion values in InvenSense Teapot demo format:
            teapotPacket[2] = fifoBuffer[0];
            teapotPacket[3] = fifoBuffer[1];
            teapotPacket[4] = fifoBuffer[4];
            teapotPacket[5] = fifoBuffer[5];
            teapotPacket[6] = fifoBuffer[8];
            teapotPacket[7] = fifoBuffer[9];
            teapotPacket[8] = fifoBuffer[12];
            teapotPacket[9] = fifoBuffer[13];
            Serial.write(teapotPacket, 14);
            teapotPacket[11]++; // packetCount, loops at 0xFF on purpose
        #endif

        // blink LED to indicate activity
        blinkState = !blinkState;
        digitalWrite(LED_PIN, blinkState);
    }
}
```

Sinamajidi, 2024-09-01:
> @Sinamajidi Ok, this information is useful to me, can you try to test this sketch, please share with me the output of the serial terminal
>
> ```
> #include "I2Cdev.h"
>
> #include [...]

I did it here is the result:

```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
Initializing I2C devices...
Testing device connections...
MPU6050 connection failedMPU6050 connection failedMPU6050 connection failedMPU6050 connection failedMPU6050 connection failedMPU6050 connection failedMPU6050 connection failed
```

sabas1080, 2024-09-06:
@Sinamajidi it seems that the mpu6050 you have is not working or you should check the connection.

Sinamajidi, 2024-09-07:
@sabas1080 

Do you mean it doesn't work at all or just for the last test result?

Bcz as you have seen, the DMP is working, so the chip should work normally despite the fact that acceleration and direction vectors can't be accessible with lib as I told you at first.

sabas1080, 2024-09-09:
@Sinamajidi 
if the sketch does not pass the connection test, you should check the mpu6050 wiring or the chip is not working.

```
if(mpu.testConnection() == False){
    while(True){
      Serial.print("MPU6050 connection failed");
      delay(500);
    }
```

Sinamajidi, 2024-10-03:
There were problems with the pins that I connected to the esp32, now I am using the rawData example and I connected the module to the esp32 default I2C pins and the problem has been solved!

closed 2024-10-03 by Sinamajidi

Sinamajidi, 2025-02-05:
@sabas1080 @Eric286 

I was uploading the following example on esp32 connected to MPU 92.65 with default i2c pins:
https://github.com/ElectronicCats/mpu6050/blob/master/examples/MPU6050_DMP6_ImuData_for_ROS/MPU6050_DMP6_ImuData_for_ROS.ino

I figured out through debugging that mpu 92.65 has a device ID of 58 while in the definition of mpu.testConnection() this number is not included thus causing it to return false. So after I defined device ID 58 beside others to be checked it worked and dump data was shown in the serial monitor.

I suggest you define a directivefor the library so whenever the user makes it 1 it includes the device ID of 58 to the return condition mpu.testConnection() definition and the rest of the other changes.

Sinamajidi, 2025-02-26:
Done!

https://github.com/ElectronicCats/mpu6050/pull/95