ElectronicCats/mpu6050 #91: Mpu6050 freezes on demo sketch

Vespertilo, 2024-11-11:
I recently purchased an MPU6050 from amazon and attempted to get it working with an Arduino Uno. I decided to use this library to get the data from the board with DMP. 

Unfortunately, I have run into the issue that it freezes whenever I run sketches utilizing DMP. I first tried the MPU6050_DMP6 example sketch that came with this library, and it works, but only for a few seconds. It will properly connect and print out values, for anywhere from 1-5 seconds, and then completely stop printing values to the serial monitor. The green LED is still on when I look at the MPU6050.

Out of curiosity, I tried the IMU_Zero example sketch which worked like a charm. I left it running for a few minutes to confirm no freezes, and none were to be found.

To rule out suspicion that it could be a faulty chip, I got the other MPU6050 of the same type that I had and tried to run it, and I ran into the exact same issues. It appears to be only an issue when using DMP. Whenever I access the raw values there are no hang ups.
(Serial Monitor While Working)
[image]

I have the MPU6050 connected to the SCL and SDA pins on the Uno. I have it connected to 5v on the Uno, and INT is connected to digital pin 2. I am using Mac OS 15.0.1, and Arduino IDE 2.3.3

Elizabeth-0, 2024-11-12:
Hi @Vespertilo,

Thank you for contacting us!

As the first approach, we would like to ask you for the version of the library you are using and confirm whether you are moving the module.

Note that the MPU6050_DMP6 example includes a function that only displays data when there are changes, in contrast to the IMU_ZERO example, which continuously shows the values.

Please feel free to let us know if you have any queries or updates.
 
Have a nice day!

Vespertilo, 2024-11-13:
Thanks for the response! I just figured out the issue so I'll say what I did here.

Essentially, it turns out that it is a problem with the chip that I bought. It was a cheap MPU6050 from amazon, and that leads to some... issues. Apparently there is a known issue with Arduinos where the I2C bus freezes when using the Wire library, and it happens with some chips and not others. Go figure 🤷 

There is a very simple solution: `Wire.setWireTimeout(3000, true);`
[image]

This fixxed all of my issues and the program ran for minutes on end without any freezes.

closed 2024-11-13 by Vespertilo

Elizabeth-0, 2024-11-14:
Thanks for your answer!

It is good news that you found a solution and thank you for letting us know about the issue with the library. 

Have a nice day and good luck with future projects!