ElectronicCats/mpu6050 #96: Missing dmpGetLinearAccelInWorld method in example

31vi5, 2025-05-08:
**Describe the bug**
When attempting to run the example code from MPU6050_DMP6 a following error is thrown

(error: 'MPU6050 {aka class MPU6050_6Axis_MotionApps20}' has no member named 'dmpGetLinearAccelInWorld'; did you mean 'dmpSendLinearAccelInWorld'?)

**To Reproduce**
Run MPU6050_DMP6 example and uncomment //#define OUTPUT_READABLE_WORLDACCEL

**Expected behavior**
Code works

**Screenshots**

**To help us to understand your situation, we would like to ask you for the following additional information::**
Arduino nano
Linux
Arduino IDE 2.3.5

**Additional context**

xpeqex, 2025-05-08:
Hey @31vi5 

Thank you for bringing this to our attention.

Seems that we did not clearly declare that to use the output format "OUTPUT_READABLE_WORLDACCEL" you must use the `MPU6050_6Axis_MotionApps612.h` header instead of the one for version 20. So please try uncommenting the code line 31 from the code and comment code line 30.

https://github.com/ElectronicCats/mpu6050/blob/8692adce4a40ad6e8b2db81d5b12586854d5895f/examples/MPU6050_DMP6/MPU6050_DMP6.ino#L30-L31

Please let us know if this solved the problem.

xpeqex, 2025-05-15:
We will close this issue now. Feel free to open a new one if you have any other issues.

closed 2025-05-15 by xpeqex