ElectronicCats/mpu6050 #97: The value obtained by getDeviceID() is 0x38

Zhentao-Lin, 2025-05-16:
Recently, the mpu6050 module I purchased found that the device ID value read was 0x38 instead of the previous 0x34. The usage of other functions is normal, except that the device ID address is different. I would like to ask you to add support for this device in the code.

bool MPU6050_Base::testConnection() {
  uint8_t deviceId = getDeviceID();
  return (deviceId == 0x34) ||  (deviceId == 0x38) ||  (deviceId == 0xC) || (deviceId == 0x3A);
}

Elizabeth-0, 2025-05-19:
Hello @Zhentao-Lin,

Thank you for bringing this to our attention. 

We appreciate you taking the time to report the issue with the MPU6050 device ID variation. 

If you are interested, please contribute to the repository by submitting a pull request. You can find our [Contribution Manual](https://github.com/ElectronicCats/electroniccats-cla/blob/main/electroniccats-contribution-manual.md) here.

Please feel free to let us know of any updates before closing this issue.

Have a nice day!✨

Elizabeth-0, 2025-05-27:
Hello @Zhentao-Lin,

Hope you are well. Just following up on this issue. Let us know if there are any updates before closing it.

Have a nice day!✨

closed 2025-05-28 by Zhentao-Lin